geopm/geopm #698: Report/trace header field "Start Time" is incorrect.

bgeltz, 2019-08-08:
I'm seeing:

```
##### geopm 1.0.0+dev102g2e5e862 #####
Start Time: Fri Jan 09 13:58:57 1970
Profile: /home/bgeltz/build/geopm/bin/geopmbench
Agent: monitor
Policy: {}
```

Clearly that's wrong.

bgeltz, 2019-08-09:
The start time is also not handled properly in the io.py's AppOutput class:

```
                                                                                        count  energy_dram  energy_pkg  frequency                  id  mpi_runtime  runtime  \
version              start_time name       agent   node_name iteration region
1.0.0+dev103gd970e99 NaN        geopmbench monitor mcfly15   1         epoch              100        0.000        0.00      0.000                None          0.0  5.83695
                                                                       unmarked-region      0      105.841     1619.71    103.326  0x00000000725e8066          0.0  5.94557
```

It should not be NaN.

bgeltz, 2019-08-09:
Once this is all fixed, the IndexTracker will also have to be updated to remove the start_time from the unique index identification:

https://github.com/geopm/geopm/blob/b8ed32a8ef52470e092a9e4c475a73f1034884fd/scripts/geopmpy/io.py#L495

bgeltz, 2019-08-09:
```
                 if self._start_time is None:
-                    match = re.search(r'^Start Time: (\S+)$', line)
+                    match = re.search(r'^Start Time: (.*)$', line)
                     if match is not None:
```

dannosliwcd, 2019-08-20:
Only the first host is in the part of the report file that contains the start of the time string, so the parser does not use that header field for any host after the first.

dianarg, 2019-11-11:
Fixed with https://github.com/geopm/geopm/commit/b16813674fd6f5977869efd436305ef783573a77

closed 2019-11-11 by dianarg