geopm/geopm #700: Update integration test module execution to print current test being executed

bakerbrandond, 2019-08-15:
User must ls -t test_integration to try and identify the currently running test.

bgeltz, 2019-08-15:
```./geopm_test_integration.py -v``` should do this already.

bakerbrandond, 2019-08-15:
hmm. lemme test with 'python . -v'.  will close if behaves as expected.

bgeltz, 2019-08-15:
Ok, that I'm not sure about.  The -v magic works with ```geopm_test_integration.py``` because it subclasses UnitTest.  UnitTest supports -v.  If the newer test Chris wrote doesn't subclass UnitTest, then we'll have to add -v support manually.

bakerbrandond, 2019-08-15:
The new test classes appear to need some work.  They are listed in the output at the end of running all the modules but no prints during their execution.