geopm/geopm #703: geopmlaunch errors when the number of physical cores is exceeded

bgeltz, 2019-08-21:
On a 44 core (2x22 core) node:

```
geopmlaunch srun -N 1 -n 45 --geopm-ctl=process --geopm-report=dgemm_report --geopm-trace=dgemm_trace -- geopmbench --no-init dgemm_config.json
...
<geopmpy.launcher> Cores cannot be shared between MPI ranks
```

```
geopmlaunch srun -N 1 -n 88 --geopm-ctl=process --geopm-report=dgemm_report --geopm-trace=dgemm_trace -- geopmbench --no-init dgemm_config.json
...
<geopmpy.launcher> Cores cannot be shared between MPI ranks
```

For GROMACS, the best performing config was to use 88 ranks and 1 thread.  This puts a rank on every physical core *and* every hyperthread.

dianarg, 2019-09-04:
How much better is this config than the best supported config?

bgeltz, 2019-09-13:
My current results are the following:

Config 1 - 88 ranks per node, 1 OMP thread
```PERFORMANCE_OPS="-dlb yes -npme 40 -dd 8 3 2"```
```
                Core t (s)   Wall t (s)        (%)
        Time:     2345.992       26.661     8799.2
                  (ns/day)    (hour/ns)
 Performance:       40.516        0.592
```

Config 2 - 44 ranks per node, 1 OMP thread
```PERFORMANCE_OPS="-dlb yes -npme 16 -dd 7 2 2"```
```
               Core t (s)   Wall t (s)        (%)
       Time:     1539.599       34.996     4399.3
                 (ns/day)    (hour/ns)
Performance:       30.867        0.778
```

I've run each config multiple times, annd the Wall t (s) is relatively stable for each.  The run-to-run variance is < 0.5 seconds.  So we're talking about a ~31% increase in runtime for this problem for not using the optimal ranks/threads params.  I haven't looked at what this does for the power consumption of the workload, but it may also be impacted.

dianarg, 2020-03-31:
Needs a supporting feature to make GEOPM runtime capable of supporting multiple ranks per core.

cmcantalupo, 2021-06-15:
This can be accomplished with the --geopm-affinity-disable option.   The only limitation is in the launch code to generate the affinity mask.

closed 2021-06-15 by cmcantalupo