geopm/geopm #745: Cannot run integration tests from within test_integration folder

dianarg, 2019-09-25:
Currently, the only way to run the integration tests is with `python -m test_integration [TestIntegration.test_name]` from the geopm source root.  The following methods within the test folder are not working:
    `python .`
    `python -m geopm_test_integration`
    `./geopm_test_integration.py`
    `python -m unittest geopm_test_integration.py`

The same error appears in different locations: `ValueError: Attempted relative import in non-package`

Physsix27, 2019-10-06:
I can give it a try to this issue.
Can yougiv me a little more details of what the problem is, do you wan't to fix `valueError: Attempted relative import in non-package` ?
Do you want to be able to run your tests from any location? What are you using for your tests, `pytest`?

cmcantalupo, 2019-10-07:
Thanks for offering to help @Physsix27.  

This issue cropped up as we are trying to get geopmpy compatible with python 3.x.  One of the requirements for 3.x compatibility is to avoid relative imports.  The patch that enforces this is here: 8a74b18cd4c0d9a7560b671e46efe9e598509441, but this resulted in the problem reported in this issue.  We would like to be able to execute the integration tests as they are documented in the readme:

https://github.com/geopm/geopm#testing

it would be for extra credit to make it work from any directory.  

We are using the python `unittest` module for our integration tests.  

Please feel free to follow up with any other questions you have.  Also feel free to request access to our slack channel at https://geopm.slack.com.  @dannosliwcd has been doing most of the python 3.0 compatibility work for us so far.

closed 2022-03-23 by cmcantalupo