geopm/geopm #764: GEOPM plugin python wrappers only able to access builtin agents

bakerbrandond, 2019-10-03:
Though geopmagent CLI is able to access plugin agents as specified by GEOPM_PLUGIN_PATH the python wrappers are not able to:
CLI:
```
$ geopmagent
monitor
power_balancer
power_governor
energy_efficient
frequency_map
example
```

Python:
```
>>> import os
>>> print(os.environ['GEOPM_PLUGIN_PATH'])
/home/bbaker1/dev/geopm/tutorial/agent
>>> num_agent = _ffi.new("int *")
>>> err = _dl.geopm_agent_num_avail(num_agent)
>>> err
0
>>> num_agent[0]
5
```

bakerbrandond, 2019-10-04:
create an integration test that adds the example plugins the plugin path and verifies the Python C-wrappers are able to list and describe the plugins.  Dependent on #396 which is tracking the fact that none of the test infrastructure currently builds the tutorials.

dannosliwcd, 2020-03-10:
https://review.gerrithub.io/c/geopm/geopm/+/475803 has a proposed solution. Alternatively, it would be nice if we could be explicit about load order.

dianarg, 2020-03-31:
Related to #986

commit 2f454ec4 (2020-04-06): Fix python cffi wrapper dlopen calls to enable plugin load

- Fixes #764

Change-Id: Ib2dfdb315f2af34e8273df7716eeadb79b3f9428
closed 2020-04-06 by cmcantalupo