geopm/geopm #769: Use locked pid file to enforce single Controller per node instead of existence of shared memory

dianarg, 2019-10-08:
Related to #766 

Currently, the existence of the shared memory region between the controller and the app is used to prevent multiple controllers from running on the same node.  This is a problem because if stale keys are left around, geopm will not start.

Instead, allow the shared memory region to be overwritten (remove O_EXCL from flags) and use a different file known only to the controller to check that a single controller is running.  A file locked with flock() should be automatically unlocked when the process holding the lock dies.

dianarg, 2020-12-02:
This may need to use `fcntl()` instead of `flock()` to deal with NFS

cmcantalupo, 2022-03-24:
Altenately, provide controller launch as a DBus call for the GEOPM Service.  This would fork a process in a similar fashion to the batch server launch.