geopm/geopm #772: Version 1.1 release checklist

dianarg, 2019-10-10:
- [ ] Examine changes to all installed headers and command line interfaces since last tag.  If any interfaces have changed, increase the minor or major version.
- [ ] Update README dates and contact info.
- [ ] Update ChangeLog based on git log.
- [ ] Update coverage numbers.
- [ ] Update doxygen, man pages and code coverage public web pages.
- [ ] Make sure OBS is building the dev branch correctly.
- [ ] Make sure static analysis is passing (IPLDT, klocwork, protex).
- [ ] Push tarball to python PIP test server and make sure it builds properly there.
- [ ] Run integration tests on mcfly, quartz, and theta.
- [ ] Make sure tutorials build and execute.

bgeltz, 2019-11-13:
re: theta integration testing

I just went back through the data from the last run I did.  It's here:
```
/projects/intel/geopm-home/shared/output/387531.error
```

I started this run from tag: v1.1.0 and added only fe468fe3.  This patch addresses integration test failures from previous jobs.  It lowers the big-o chosen for certain tests to avoid OOM issues.  I have another patch up 3a77709f to fix issues with running with the installed module file.  These need to be included in the geopm-theta tutorial README as well.

This is looking pretty good.  36 tests were run, 8 were skipped and 26 passed clean (including OMPT support).

Here's the list of skipped tests:
```
64:test_agent_energy_efficient (test_integration.geopm_test_integration.TestIntegration) ... skipped 'Requires SLURM batch session.'
111:test_power_balancer (test_integration.geopm_test_integration.TestIntegration) ... skipped 'Requires SLURM batch session.'
132:test_report_and_trace_generation_application (test_integration.geopm_test_integration.TestIntegration) ... skipped 'ALPS does not support multi-application launch on the same nodes.'
133:test_report_and_trace_generation_pthread (test_integration.geopm_test_integration.TestIntegration) ... skipped 'This test only enabled on known working systems.'
134:test_report_generation_all_nodes (test_integration.geopm_test_integration.TestIntegration) ... skipped "Requires non-sbatch SLURM session for alloc'd and idle nodes."
166:test_geopmread_all_signal_agg (test_integration.geopm_test_integration.TestIntegrationGeopmio) ... skipped 'Requires SLURM batch session.'
169:test_geopmread_signal_value (test_integration.geopm_test_integration.TestIntegrationGeopmio) ... skipped 'Requires SLURM batch session.'
171:test_geopmwrite_set_freq (test_integration.geopm_test_integration.TestIntegrationGeopmio) ... skipped 'Requires SLURM batch session.'
```

These are all expected at the moment.  I have a branch on my fork where I've been working on trying to run the skipped tests and have been making headway.  The work is on the ```bgeltz-theta-tests``` branch here:
https://github.com/bgeltz/geopm/tree/bgeltz-theta-tests

The batch test decorator is specific to SLURM, which is why you see so many tests skipped because of it.  There's a patch on my fork to add ALPS support (23c45ceb).  Running with that patch, and you'll get the next set of problems:

The geopmread/write tests are skipped because they're trying to execute the ```geopmread/write``` binaries on the MOM nodes.  They should be attempting this through the launcher instead of directly from the test code to ensure this happens on compute nodes.  The tools work OK on the computes, the test code just needs to be updated.

```test_report_and_trace_generation_pthread``` should work with ALPS.  I'll try to update the test decorator and re-run.

```test_power_balancer``` fails because it has a recently added dependency on getting the allocated nodes for the job, but that dependency was only coded up for SLURM.  ALPS support is available, and needs to be added.  Also, as it is coded the function is not returning the information the test is expecting.  It's providing the test with all the allocated nodes for the cluster, not just the allocated nodes for the job.  This also must be fixed.

```test_agent_energy_efficient``` needs to be re-evaluated with these fixes as well.

bgeltz, 2019-11-20:
I've run the tests on Theta and Quartz.  I'm tracking the Theta work in #802.  The results on Quartz are as follows:
```
======================================================================
FAIL: test_agent_energy_efficient (test_integration.geopm_test_integration.TestIntegration)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/p/lustre2/geltz2/git/geopm/test_integration/geopm_test_integration.py", line 1249, in test_agent_energy_efficient
    self.assertLess(-0.1, runtime_savings_epoch)  # want -10% or better
AssertionError: -0.1 not less than -0.10435035997608538

======================================================================
FAIL: test_monotone_frequency (test_integration.test_ee_stream_dgemm_mix.TestIntegrationEEStreamDGEMMMix)
Test that agent selects lower frequencies for regions with more
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/p/lustre2/geltz2/git/geopm/test_integration/test_ee_stream_dgemm_mix.py", line 149, in test_monotone_frequency
    msg='Chosen frequency decreased with increasing dgemm fraction')
AssertionError: Chosen frequency decreased with increasing dgemm fraction

======================================================================
FAIL: test_geopmwrite_set_freq (test_integration.geopm_test_integration.TestIntegrationGeopmio)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/p/lustre2/geltz2/git/geopm/test_integration/geopm_test_integration.py", line 1479, in test_geopmwrite_set_freq
    self.assertEqual(max_freq, result)
AssertionError: 2100000000.0 != 1200000000.0

----------------------------------------------------------------------
Ran 36 tests in 2694.152s

FAILED (failures=3, skipped=2)
Job 3963286 finished at 2019-11-19_1820.15...
Job 3963286 complete!
```

The EE agent failure looks to be that the threshold for performance was narrowly missed on this set of nodes.  I'm going to re-run later and see if I can get it to pass.

I've seen this ```test_monotone_frequency``` failure sporadically on every machine I've tested on lately.  @bakerbrandond did some digging and found that for a given stream/DGEMM configuration, the neighboring configurations are sometimes off by 1 P-State.  More digging is necessary to bottom out on if there is a real problem causing this or if the test is too strict. 

I'm going to open a new issue about ```test_monotone_frequency``` failing intermittently.

I'm not sure what's going on with the ```set_freq``` test.  I ran on Quartz from the tip of dev, which has the ```geopmread/write``` calls happening directly out of the test launcher.  This is a problem on ALPS, but should be fine here.  More debug is necessary.

bgeltz, 2019-11-21:
It turns out ```set_freq``` is issuing writes then reads too quickly for the system to respond.  I also saw this issue on Theta, and the fix is a part of the commit I'm already working on here:
https://github.com/bgeltz/geopm/blob/3e62bf7ee9124f9a8dd7f469d0858268f6927ddd/test_integration/geopm_test_integration.py#L1452

closed 2019-11-25 by cmcantalupo