geopm/geopm #782: Run AMG with no markup with EE agent

dianarg, 2019-10-16:


dianarg, 2019-10-16:
Here's what I have recorded for valid (not necessarily great) input parameters:
```
        if self.num_nodes == 128 and self.ranks_per_node == 16:
            self.app_params = '-problem 1 -n 96 96 96 -P 16 16 8'  # total product of -P must == total ranks
        elif self.num_nodes == 1 and self.ranks_per_node == 16:
            self.app_params = '-problem 1 -n 96 96 96 -P 4 2 2'
        else:
            raise RuntimeError("No input size defined for amg on {} nodes with {} ranks per node".format(self.num_nodes, self.ranks_per_node))
```

dannosliwcd, 2020-01-22:
Closing since we have run AMG2013 with GEOPM, without markup, and broken down this issue into additional stories.

Ran with https://computing.llnl.gov/projects/co-design/amg2013

The summary of results is that this benchmark demonstrates a case where GEOPM can be enhanced to perform better on short-running regions or infrequent long-running regions. #855 has been opened to help with some of those cases.

AMG doesn't have complicated build steps, and its workload sizes easily scale with the number of processes made available to it (via command line options). The run time is relatively short even as problem size scales up. Size depends on the sub-benchmark, but generally includes a component that scales the data size, and another component that scales both data size and the division across processes. Due to its ease of building and running, it would be a good candidate as an external benchmark to run in automated tests. #807 has been opened to track the addition of new tests.

closed 2020-01-22 by dannosliwcd