geopm/geopm #790: Add separate parameter to agent policy for static control

cmcantalupo, 2019-10-23:
Have a parameter that explicitly controls the static policy for each built in agent that can be modified independently from the policy used by the GEOPM runtime.

dianarg, 2019-10-23:
Related to #780

dianarg, 2019-10-26:
I propose separating the policy for `enforce_policy()` entirely from the policy used by the Controller.  In other words, rather than adding a new policy parameter to each agent, instead add a different policy vector entirely (and corresponding environment variable).  The main problem is that to be able to support this feature (#780) , the set of policy parameters used at runtime must be completely disjoint from the set used for static policies.  This means that any method of the agent that takes a policy vector as input is expected to ignore part of the input.
- `adjust_platform()` is only called by the Controller at runtime; it should ignore the static policy parameters.
- `split_policy()` is only called by the Controller at runtime; it's not clear what it should pass down to children for the static policy, since the leaf agents are supposed to ignore those parameters.
- `enforce_policy()` is only called in the static policy context (`geopm_agent_enforce_policy()`); it should ignore the dynamic policy parameters.
- `validate_policy()` is called in both cases, and needs to know how to set defaults and check bounds for both sets of policy parameters.  Probably these checks will be the same for both, but at the moment we cannot rely on `validate_policy()` anyway because the implementation is incomplete or scattered in other parts of the code for some of the agents.

Instead of trying to update the existing methods used by the controller at runtime, it makes more sense to me to remove the call to `validate_policy()` from the static policy enforcement path, and handle all the checks inside `enforce_policy()`.  Maybe there are some helpers for checking hardware bounds or setting defaults that could be factored out to reduce the amount of repeated code.

One problem with this proposal is that we don't have a method like `policy_names()` in the Agent interface to get the names of static policy parameters, unless they are part of the same policy.  It would probably not be hard to add a `static_policy_names()` method, but that seems risky so close to a release.  `policy_names()` is used both in parsing the JSON input files and in the output produced by the `geopmagent` tool.

commit 658993a1 (2019-11-05): Add fixed frequency policy for EnergyEfficientAgent

- The EE agent only supports a separate static policy.
- This feature will be separated from the policy vector
  after v1.1.
- Resolves #780
- Resolves #790

Change-Id: I3e8fdb3a5568199abb414d16355e1614ed8fc821
closed 2019-11-05 by cmcantalupo