geopm/geopm #798: Clean up unclosed file warnings from python3 in integration tests

dianarg, 2019-11-08:
When running the integration tests with python 3, there are many Resource Warnings from the subprocess module like the following:
```
/home/drguttma/geopm/test_integration/geopm_test_integration.py:1346: ResourceWarning: unclosed file <_io.BufferedReader name=4>
  self.check_no_error([sig.decode(), 'board', '0'])
/usr/lib64/python3.6/subprocess.py:786: ResourceWarning: subprocess 232800 is still running
  ResourceWarning, source=self)

```

udaykiranreddykondreddy, 2019-11-08:
@dianarg can I work on this?

dianarg, 2019-11-09:
@udaykiranreddykondreddy Go right ahead!  Contributions are always welcome.  Let us know if you have any questions.

udaykiranreddykondreddy, 2019-11-09:
@dianarg Is there any platform to discuss with you or with the team?

dianarg, 2019-11-11:
You can request access to our Slack here: https://geopm.slack.com/

udaykiranreddykondreddy, 2019-11-11:
@dianarg I don't have an account ending with @intel.com, @anl.gov, @alcf.anl.gov, @llnl.gov, @sandia.gov, @in.tum.de, @tum.de, @lrz.de, @vsb.cz, @unibo.it, @cines.fr, or @genci.fr

dianarg, 2019-11-11:
Sorry about that.  You can post any questions on this issue and we'll try to help.  It will hard to test this without having run GEOPM once, so I think that's a good place to start.  Do you have access to a server where GEOPM is already installed, or where you can install it and all the dependencies?  In particular, you will need [msr-safe](https://github.com/LLNL/msr-safe/), which you need root access to set up.