geopm/geopm #817: Integration tests clean up output files even on failure

dianarg, 2019-11-25:
If GEOPM_KEEP_FILES is not set, files will be cleaned up regardless of the test outcome.  This is because `sys.exc_info() == (None, None, None)` is always true in the tearDown context in Python3.  See https://stackoverflow.com/questions/4414234/getting-pythons-unittest-results-in-a-teardown-method#comment78911300_4414788

A workaround is to always set GEOPM_KEEP_FILES and manually remove the test output files if not needed.

bgeltz, 2019-11-25:
Add'fl info about using sys.exc_clear():
https://cosmicpercolator.com/2016/01/13/exception-leaks-in-python-2-and-3/

cmcantalupo, 2020-06-12:
If we change the default we need to update the documentation here:

https://github.com/geopm/geopm/tree/dev/test_integration#geopm_keep_files

cmcantalupo, 2020-07-10:
The proposal is to always keep files, and have the nightly script take on the responsibility of cleaning up after itself, thus removing the feature requirement from integration tests.

closed 2020-08-26 by bakerbrandond