geopm/geopm #825: Create new agent that tunes frequency for entire epoch

cmcantalupo, 2019-12-03:
First attempt at this will apply a similar algorithm to the EE agent, except on the epoch rather than per region.  May be a feature of the EE agent in the future.  This task is intended to explore the effectiveness of this type of algorithm on benchmarks of interest.

eastep, 2019-12-03:
Would be better to tune frequency for long MPI calls separately from the non-MPI portion of the epoch. One heuristic and simple way to do it is to learn the best frequency over the whole epic but while always running the long MPI calls at min frequency instead of the learned value.

eastep, 2019-12-03:
Having said all of that, we really should FIRST test the existing EE Agent algorithm that we already wrote, debugged, and tuned while temporarily turning off OpenMP parallel region detection so that the EE Agent focuses on tuning frequency for long MPI calls. That approach is proven in the literature to provide significant energy savings (see the work of Cesarini, Bartolini, et al. from UniBo on COUNTDOWN).

bakerbrandond, 2019-12-03:
> Would be better to tune frequency for long MPI calls separately from the non-long-MPI-call portion of the epoch. One heuristic and simple way to do it is to learn the best frequency over the whole epoch but always run the long MPI calls at min frequency (i.e. override the learned value). The advantage of this slightly more sophisticated approach (vs. applying the same frequency to the whole epoch) is that it could still be considered a phase-adaptive approach which would distinguish it from other approaches deployed in production systems.

That has already been discussed and has already been taken into account in the current design we are implementing, which isn't tracked here in the issue.

> Having said all of that, we really should FIRST test the existing EE Agent algorithm that we already wrote, debugged, and tuned while temporarily turning off OpenMP parallel region detection so that the EE Agent focuses on tuning frequency for long MPI calls. A similar but distinct approach has already been proven in the research literature to provide significant energy savings (see the work of Cesarini, Bartolini, et al. from UniBo on COUNTDOWN).

We have discussed this as well (in the last review meeting).  Nekbone (and I believe AMG as well) require OMPT be disabled in order to observe user defined regions at all and the way that Nekbone is currently marked up even user defined regions are running too fast under the default sampling rate, which leaves the agent to only focus on long running MPI regions.  There has been discussion wrt making OMPT detection runtime configurable as well as adding more sophisticated logic to the EE agent wrt short running OMPT detected regions, though prioritization of continued discussion and much less implementation has yet to be discussed.

eastep, 2019-12-04:
> > > Would be better to tune frequency for long MPI calls separately from the non-long-MPI-call portion of the epoch. One heuristic and simple way to do it is to learn the best frequency over the whole epoch but always run the long MPI calls at min frequency (i.e. override the learned value). The advantage of this slightly more sophisticated approach (vs. applying the same frequency to the whole epoch) is that it could still be considered a phase-adaptive approach which would distinguish it from other approaches deployed in production systems.
> > 
> > That has already been discussed and has already been taken into account in the current design we are implementing, which isn't tracked here in the issue.
> 
> Good that you're already thinking along the same lines. 
> 
> > > Having said all of that, we really should FIRST test the existing EE Agent algorithm that we already wrote, debugged, and tuned while temporarily turning off OpenMP parallel region detection so that the EE Agent focuses on tuning frequency for long MPI calls. A similar but distinct approach has already been proven in the research literature to provide significant energy savings (see the work of Cesarini, Bartolini, et al. from UniBo on COUNTDOWN).
> > 
> > We have discussed this as well (in the last review meeting). Nekbone (and I believe AMG as well) require OMPT be disabled in order to observe user defined regions at all and the way that Nekbone is currently marked up even user defined regions are running too fast under the default sampling rate, which leaves the agent to only focus on long running MPI regions. There has been discussion wrt making OMPT detection runtime configurable as well as adding more sophisticated logic to the EE agent wrt short running OMPT detected regions, though prioritization of continued discussion and much less implementation has yet to be discussed.
>
> Yes it was discussed, and my point in writing about it here too is that this task seems to propose researching new algorithms in this sprint when we discussed instead focusing on evaluation of the EE Agent algorithm we already have but limiting its scope to tuning frequency in long MPI calls.