geopm/geopm #872: RawReport.region_hash may fail when one region name is a prefix for another region name

dannosliwcd, 2020-01-22:
The wrong region hash may be provided by `RawReport.region_hash` if the requested region's name matches the start of another region's name that occurs earlier in the report file.

On the line in io.py:
```if xx.startswith('Region {}'.format(region_name)):```

Example failure case:
`RawReport.region_hash()` may unexpectedly return the hash of region `myregion_1` when another region `myregion_123` was requested if `myregion_1` occurs first in the report file (e.g. if it has a greater run time)

dannosliwcd, 2020-01-22:
One solution may be to add a space after the expected region name. E.g.

```if xx.startswith('Region {} '.format(region_name)):```

sidjana, 2020-03-12:
The bug is fixed. Regression test needs to be added. RawReport needs tests.

cmcantalupo, 2020-03-12:
fix is here:
https://github.com/geopm/geopm/commit/3c8d5afe1db3273892c87224351503c0ffdf180f
@dannosliwcd does that makes sense to you?  We need some unit tests to cover this.

dannosliwcd, 2020-03-12:
@cmcantalupo Yeah that looks like it should work.

closed 2020-03-19 by cmcantalupo