geopm/geopm #886: Write prerequisites section for admin tutorial

cmcantalupo, 2020-01-23:
> estimate 2

Most documentation is covered top level README, this work is primarily about writing the test script.

Script below may be useful.
https://github.com/geopm/endeavor-tutorial/blob/master/test-msr-safe.sh

Parent story #806

bgeltz, 2020-02-03:
Filed #907 as a follow-on task.

bgeltz, 2020-02-03:
Re: Done Criteria, #2:
Since most of the things that are examined by the test script are not easily configurable by a script, and that they could touch critical bits of the OS image (e.g. the kernel command line) this line item was not attempted.  Instead pointers to the existing documentation and clear error messages when a misconfiguration is detected are provided.

PR #908 (2020-02-04): Add pre-req checker script and associated documentation
Part of story #806.

- Fixes #886.
commit b937fc78 (2020-02-04): Add prerequisite test script

- Fixes #886.

Change-Id: I4cefb88380e5fc82a665f17a4d6782b559554553
commit ecd327af (2020-02-04): Add prerequisite test script

- Fixes #886.

Change-Id: I4cefb88380e5fc82a665f17a4d6782b559554553
closed 2020-02-04 by bgeltz