geopm/geopm #905: Missing fortran compiler not detected by configure

dianarg, 2020-02-03:
If a system has no Fortran compiler, FC will be empty, causing errors when compiling geopm.f90.  The configure step still passes with return code 0 and ENABLE_FORTRAN is still set to true for the make step.

Might be related to #725

bakerbrandond, 2020-08-13:
Various attempts to test the FC var for empty string have been fruitless.

gfortran is defaulted to at the end of configure step, when FC is not present or is empty.

cmcantalupo, 2020-08-13:
The bug states that FC is empty if a fortran compiler is not in the path but --disable-fortran is not provided.  Are you not able to reproduce?

bakerbrandond, 2020-08-13:
I am able to reproduce.

The 'testing' I am referring to is in the context of the configure script with tests like:
'''if test -z "$FC"; then'''

cmcantalupo, 2020-08-13:
```
cmcantal@centos-7:~/Git/geopm (dev)]$ ./configure  --prefix=$HOME/build/geopm  --with-python=python2 
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for mpicc... /opt/ohpc/pub/mpi/openmpi-gnu7/1.10.7/bin/mpicc
Checking whether /opt/ohpc/pub/mpi/openmpi-gnu7/1.10.7/bin/mpicc responds to '-showme:compile'... yes
mpi libs: -lmpi
checking that we can build MPI programs with C... yes
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking for mpicxx... /opt/ohpc/pub/mpi/openmpi-gnu7/1.10.7/bin/mpicxx
Checking whether /opt/ohpc/pub/mpi/openmpi-gnu7/1.10.7/bin/mpicxx responds to '-showme:compile'... yes
mpi libs: -lmpi_cxx  -lmpi
checking that we can build MPI programs with C++... yes
checking for gfortran... no
checking for g95... no
checking for xlf95... no
checking for f95... no
checking for fort... no
checking for ifort... no
checking for ifc... no
checking for efc... no
checking for pgfortran... no
checking for pgf95... no
checking for lf95... no
checking for ftn... no
checking for nagfor... no
checking for xlf90... no
checking for f90... no
checking for pgf90... no
checking for pghpf... no
checking for epcf90... no
checking for g77... no
checking for xlf... no
checking for f77... no
checking for frt... no
checking for pgf77... no
checking for cf77... no
checking for fort77... no
checking for fl32... no
checking for af77... no
checking whether we are using the GNU Fortran compiler... no
checking whether  accepts -g... no
checking for mpif95... no
checking for mpif90... /opt/ohpc/pub/mpi/openmpi-gnu7/1.10.7/bin/mpif90
Checking whether /opt/ohpc/pub/mpi/openmpi-gnu7/1.10.7/bin/mpif90 responds to '-showme:compile'... yes
mpi libs: -lmpi_usempif08  -lmpi_usempi_ignore_tkr  -lmpi_mpifh  -lmpi
checking that we can build MPI programs with Fortran... no
checking for g77... no
checking for xlf... no
checking for f77... no
checking for frt... no
checking for pgf77... no
checking for cf77... no
checking for fort77... no
checking for fl32... no
checking for af77... no
checking for xlf90... no
checking for f90... no
checking for pgf90... no
checking for pghpf... no
checking for epcf90... no
checking for gfortran... no
checking for g95... no
checking for xlf95... no
checking for f95... no
checking for fort... no
checking for ifort... no
checking for ifc... no
checking for efc... no
checking for pgfortran... no
checking for pgf95... no
checking for lf95... no
checking for ftn... no
checking for nagfor... no
checking whether we are using the GNU Fortran 77 compiler... no
checking whether  accepts -g... no
checking for mpif77... /opt/ohpc/pub/mpi/openmpi-gnu7/1.10.7/bin/mpif77
Checking whether /opt/ohpc/pub/mpi/openmpi-gnu7/1.10.7/bin/mpif77 responds to '-showme:compile'... yes
mpi libs: -lmpi_usempif08  -lmpi_usempi_ignore_tkr  -lmpi_mpifh  -lmpi
checking that we can build MPI programs with Fortran 77... no
configure: Building with MPI C support
configure: Building with MPI C++ support
configure: error: Failed to determine MPI Fortran build flags use --with-mpi-bin or --with-mpicxx or --disable-mpi

```

cmcantalupo, 2020-08-13:
If mpi is enabled, and the fortran compiler is not in my path, I see that error.  If mpi is not enabled, fortran is not used.

bakerbrandond, 2020-08-14:
Fixed by #725

closed 2020-08-14 by bakerbrandond