geopm/geopm #971: Own and drive checkpoint 8 deliverables on 4 node Skylake cluster

bakerbrandond, 2020-03-04:
Done criteria:  System installed GEOPM and validated as measured by passing integration tests.
Pre-requisite dependent software configuration has been validated.

bakerbrandond, 2020-03-09:
System specific RAPL enablement reference [here](https://wiki.ith.intel.com/display/P3Wiki/Cluster+Maintenance#ClusterMaintenance-EnablingRAPL)

bakerbrandond, 2020-03-09:
https://github.com/geopm/geopm/pull/984

updated README.md PR

sidjana, 2020-03-10:
This is a very hands-on task. This also includes back and forth communication with the system-admin of cluster-C. Steps include: (a) requesting a user account from the system-admin of cluster-C (b) building and installing a GEOPM RPM on the machine. (c) documentation of challenges due to missing packages/compilers/libraries, missing HW support, etc. (d) communicating to the system admin about these issues and presenting the contents of the said documentation