geopm/geopm #976: Ensure that the integration tests pass

sidjana, 2020-03-04:


bgeltz, 2020-03-10:
WIP branch:
https://github.com/bgeltz/geopm/tree/bgeltz-fix-int-tests

That branch has 2 required patches to get the tests going.  The first fixes a legitimate issue with unicode string handling in the launcher.py that took a long while to sort out (still not sure if this is how we want it).  The other sets the integration tests up to use the monitor instead of the power_governor.  I don't know that we want that change in dev, so I'm holding off on creating a PR for now.

Here's the summary of the first pass at this:
```
Ran 44 tests in 345.584s
FAILED (failures=2, errors=8, skipped=17) 
```

Here's list of failed tests:
```
======================================================================
ERROR: test_agent_energy_efficient_single_region (test_integration.geopm_test_integration.TestIntegration)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/geopm_test_integration.py", line 1196, in test_agent_energy_efficient_single_region
    self.assertEqual(region['requested-online-frequency'], min_freq, msg=msg)  # freq should reduce
KeyError: 'requested-online-frequency'

======================================================================
ERROR: test_report_and_trace_generation_application (test_integration.geopm_test_integration.TestIntegration)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/geopm_test_integration.py", line 246, in test_report_and_trace_generation_application
    launcher.run(name)
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/geopm_test_launcher.py", line 236, in run
    launcher.run(stdout=outfile, stderr=outfile)
  File "/dss/dsshome1/07/di57fof2/git/geopm/scripts/geopmpy/launcher.py", line 555, in run
    raise subprocess.CalledProcessError(pid.returncode, argv_mod)
CalledProcessError: Command 'srun -N 4 -n 16 --cpu-bind v,mask_cpu:0x1ffc,0xffe000,0x1ffc000000,0xffe000000000 -I30 -t 10 -J test_report_and_trace_generation_application --cpu-fr
eq=Performance -- geopmbench --verbose test_report_and_trace_generation_application_app.config' returned non-zero exit status 1

======================================================================
ERROR: setUpClass (test_integration.test_omp_outer_loop.TestIntegrationOMPOuterLoop)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/test_omp_outer_loop.py", line 111, in setUpClass
    launcher.run(curr_run)
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/geopm_test_launcher.py", line 236, in run
    launcher.run(stdout=outfile, stderr=outfile)
  File "/dss/dsshome1/07/di57fof2/git/geopm/scripts/geopmpy/launcher.py", line 555, in run
    raise subprocess.CalledProcessError(pid.returncode, argv_mod)
CalledProcessError: Command 'srun -N 1 -n 5 --cpu-bind v,mask_cpu:0x2,0x1ffc,0xffe000,0x1ffc000000,0xffe000000000 -I30 -t 100 -J test_omp_outer_loop_with_ompt --cpu-freq=Performance -- /dss/dsshome1/07/di57fof2/git/geopm/test_integration/.libs/test_omp_outer_loop --verbose' returned non-zero exit status 1

======================================================================
ERROR: test_energy_efficient_max_freq (test_integration.test_static_policy.TestIntegrationStaticPolicy)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/test_static_policy.py", line 86, in setUp
    geopm_test_launcher.geopmwrite("POWER_PACKAGE_LIMIT board 0 {}".format(self._tdp_power))
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/geopm_test_launcher.py", line 109, in geopmwrite
    raise err
CalledProcessError: Command 'srun -N 1 -n 1 -J geopm_allocation_test --cpu-freq=Performance -- geopmwrite POWER_PACKAGE_LIMIT board 0 240.0' returned non-zero exit status 22

======================================================================
ERROR: test_freq_map_max_freq (test_integration.test_static_policy.TestIntegrationStaticPolicy)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/test_static_policy.py", line 86, in setUp
    geopm_test_launcher.geopmwrite("POWER_PACKAGE_LIMIT board 0 {}".format(self._tdp_power))
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/geopm_test_launcher.py", line 109, in geopmwrite
    raise err
CalledProcessError: Command 'srun -N 1 -n 1 -J geopm_allocation_test --cpu-freq=Performance -- geopmwrite POWER_PACKAGE_LIMIT board 0 240.0' returned non-zero exit status 22

======================================================================
ERROR: test_monitor_no_policy (test_integration.test_static_policy.TestIntegrationStaticPolicy)
----------------------------------------------------------------------
Traceback (most recent call last):
[... 41 lines cut ...]
Traceback (most recent call last):
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/geopm_test_integration.py", line 1158, in test_agent_frequency_map_policy
    self._test_agent_frequency_map('test_agent_frequency_map_policy', use_env=False)
[... cut ...]
```test_omp_outer_loop.py``` looks to be missing OMPT support.  The compiler is a bit older, so I fear we're missing the hooks we need for automatic support.

```test_energy_efficient_max_freq``` , ```test_monitor_no_policy```, and ```test_freq_map_max_freq``` are trying to write to the RAPL MSR and should not be.

```test_agent_frequency_map_env``` and ```test_agent_frequency_map_policy``` are pretty far off the mark.  I suspect this has something to do with the current pstate driver. More debug is necessary.

I'm going to change a few things and rerun with the verbose flag to get the list of skipped tests next.

bgeltz, 2020-03-11:
I was able to resolve the OMPT issues by using a newer compiler.  We'll need to document this in the tutorial collateral for this system.

The EE agent is struggling with this system because of the delta between the maximum frequency reported by sysfs/cpuinfo and the actual default operating frequency.  There is a mismatch.  I've altered some of the tests to just use the value in IA32_PERF_CTL as the "max_freq" to start learning from.  This gets some to pass but exposes the next bug.

Trying to run at Pn results in a higher than expected frequency in the reports.  I'm using the frequency_map agent to try to pin the frequency at Pn, but the reports are showing that the regions ran at somewhere between Pn-1 and Pn-2.  I still suspect frequency governor is doing something wonky.

bgeltz, 2020-03-11:
The delta between requested frequency and achieved frequency holds true at Pn-2.  Requesting Pn-2 results in an achieved frquency of Pn-4.

bgeltz, 2020-03-11:
I just re-ran the integration tests with the above fixes.  Here's the current summary:
```
----------------------------------------------------------------------
Ran 46 tests in 314.678s

FAILED (failures=2, errors=7, skipped=17)
```

```
======================================================================                                                                                                  [103/8791]
ERROR: test_agent_energy_efficient_single_region (test_integration.geopm_test_integration.TestIntegration)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/geopm_test_integration.py", line 1198, in test_agent_energy_efficient_single_region
    self.assertEqual(region['requested-online-frequency'], min_freq, msg=msg)  # freq should reduce
KeyError: 'requested-online-frequency'

======================================================================
ERROR: test_report_and_trace_generation_application (test_integration.geopm_test_integration.TestIntegration)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/geopm_test_integration.py", line 248, in test_report_and_trace_generation_application
    launcher.run(name)
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/geopm_test_launcher.py", line 237, in run
    launcher.run(stdout=outfile, stderr=outfile)
  File "/dss/dsshome1/07/di57fof2/git/geopm/scripts/geopmpy/launcher.py", line 557, in run
    raise subprocess.CalledProcessError(geopm_pid.returncode, geopm_argv)
CalledProcessError: Command 'srun -N 4 -n 4 --cpu-bind v,mask_cpu:0x2 -I30 -t 10 -J test_report_and_trace_generation_application --cpu-freq=Performance geopmctl' returned non-zer
o exit status 1

======================================================================
ERROR: test_energy_efficient_max_freq (test_integration.test_static_policy.TestIntegrationStaticPolicy)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/test_static_policy.py", line 86, in setUp
    geopm_test_launcher.geopmwrite("POWER_PACKAGE_LIMIT board 0 {}".format(self._tdp_power))
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/geopm_test_launcher.py", line 109, in geopmwrite
    raise err
CalledProcessError: Command 'srun -N 1 -n 1 -J geopm_allocation_test --cpu-freq=Performance -- geopmwrite POWER_PACKAGE_LIMIT board 0 240.0' returned non-zero exit status 22

======================================================================
ERROR: test_freq_map_max_freq (test_integration.test_static_policy.TestIntegrationStaticPolicy)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/test_static_policy.py", line 86, in setUp
    geopm_test_launcher.geopmwrite("POWER_PACKAGE_LIMIT board 0 {}".format(self._tdp_power))
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/geopm_test_launcher.py", line 109, in geopmwrite
    raise err
CalledProcessError: Command 'srun -N 1 -n 1 -J geopm_allocation_test --cpu-freq=Performance -- geopmwrite POWER_PACKAGE_LIMIT board 0 240.0' returned non-zero exit status 22

======================================================================
ERROR: test_monitor_no_policy (test_integration.test_static_policy.TestIntegrationStaticPolicy)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/test_static_policy.py", line 86, in setUp
    geopm_test_launcher.geopmwrite("POWER_PACKAGE_LIMIT board 0 {}".format(self._tdp_power))
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/geopm_test_launcher.py", line 109, in geopmwrite
    raise err
CalledProcessError: Command 'srun -N 1 -n 1 -J geopm_allocation_test --cpu-freq=Performance -- geopmwrite POWER_PACKAGE_LIMIT board 0 240.0' returned non-zero exit status 22

======================================================================
ERROR: test_power_balancer_power_cap (test_integration.test_static_policy.TestIntegrationStaticPolicy)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/test_static_policy.py", line 86, in setUp
    geopm_test_launcher.geopmwrite("POWER_PACKAGE_LIMIT board 0 {}".format(self._tdp_power))
  File "/dss/dsshome1/07/di57fof2/git/geopm/test_integration/geopm_test_launcher.py", line 109, in geopmwrite
    raise err
CalledProcessError: Command 'srun -N 1 -n 1 -J geopm_allocation_test --cpu-freq=Performance -- geopmwrite POWER_PACKAGE_LIMIT board 0 240.0' returned non-zero exit status 22

[... 38 lines cut ...]
test_power_balancer (test_integration.geopm_test_integration.TestIntegration) ... skipped 'Define GEOPM_RUN_LONG_TESTS in your environment to run this test.'
test_power_consumption (test_integration.geopm_test_integration.TestIntegration) ... skipped 'Define GEOPM_RUN_LONG_TESTS in your environment to run this test.'
test_region_runtimes (test_integration.geopm_test_integration.TestIntegration) ... skipped 'Define GEOPM_RUN_LONG_TESTS in your environment to run this test.'
[... cut ...]
I think I understand all of these failures now.  The EE agent never finishes learning because of a combination of 2 things:
1. The delta in actual sticker frequency vs. reported sticker frequency.
2. The delta between achieved frequency and requested frequency.

The other skip tests that require the ```RUN_LONG_TESTS``` option won't provide any new information since the short tests are failing.  We need to fix the short ones first.

At this point, I think I have the information I need to open new issues for a future investigation.

bgeltz, 2020-03-12:
In trying to characterize what the problem is with frequency control, I've discovered a few ostensibly related tidbits:

When I allocated a node for testing, it is released to me with the userspace governor set.  If I try to set frequencies with geopmwrite, run a small job, then use geopmread to verify the same frequency is still in PERF_CTL, they match.

I am able to change the governor to performance using the srun option (which is interesting because doing the same through cpupower is forbidden unless you are root).  If I do that, and re-run the previous experiment, I see the PERF_CTL jump to the max turbo frequency 100 % of the time.  Test example:

```
[di57fof2@i01r01c01s01 test_integration]$ srun -N 1 -n 1 --cpu-freq=Performance -- hostname
i01r01c01s01
[di57fof2@i01r01c01s01 test_integration]$ geopmread MSR::PERF_CTL:FREQ core 0
2300000000
[di57fof2@i01r01c01s01 test_integration]$ srun -N1 -n1 hostname
i01r01c01s01
[di57fof2@i01r01c01s01 test_integration]$ geopmread MSR::PERF_CTL:FREQ core 0
3900000000
```

While this is weird, I'm not sure how related it is to the achieved frequency being greater than the requested frequency issue I described in my previous comments.  I changed the launcher to emit "userspace" instead of "performance" since I know that "userspace" does not override the PERF_CTL value.  I re-ran the frequency map integration test with fixed frequencies for all the regions.  I performed a few experiments at P0, P1, Pn -1, and Pn.  In *all* cases the achieved frequency was greater than the requested frequency (even after ensuring the "userspace" governor was in place.

The next step here would be to examine the trace file and plot achieved frequency and requested frequency over time.

Some useful commands:
```
GEOPM_KEEP_FILES=yes ./geopm_test_integration.py -v TestIntegration.test_agent_frequency_map_policy

geopmread MSR::PERF_CTL:FREQ board 0
geopmwrite MSR::PERF_CTL:FREQ board 0 2300000000

srun -N 1 -n 1 --cpu-freq=Performance -- hostname

cpupower frequency-info
```

Suffice to say there's a lot wrong with our ability to control frequency at the moment.  I'm going to start opening issues and stop digging now.

bgeltz, 2020-03-12:
I've opened #990 to continue exploring the frequency control issues.  This work will be of the highest importance for this cluster given the intended use case.

After that #991 describes how we'll need to fix up the integration tests for this cluster.

closed 2020-03-12 by bgeltz