geopm/geopm #980: Cryptic error messages with debug builds of geopmread

bgeltz, 2020-03-05:
When running ```geopmread``` with a debug build, using the wrong domain or the wrong domain index for a given signal results in a very cryptic error message.  The following cases result in the same error:

1. Using an index that is out-of-bounds for a given domain results in the MPIComm error.
```geopmread POWER_DRAM board_memory 2```
2. Using a domain that is too granular for the given signal results in the MPIComm error.
```geopmread POWER_DRAM core 0```

```
terminate called after throwing an instance of 'geopm::Exception'
  what():  <geopm> Invalid argument: PluginFactory::make_plugin(): name: "MPIComm" has not been previously registered: at geopm/src/PluginFactory.hpp:85
Aborted
```

With release builds, you see the proper error message:
```
$ geopmread POWER_DRAM core 0
Error: cannot read signal: <geopm> Invalid argument: PlatformIOImp::read_signal(): domain 2 is not valid for signal "ENERGY_DRAM": at geopm/src/PlatformIO.cpp:575

$ geopmread POWER_DRAM board_memory 2
Error: cannot read signal: <geopm> Invalid argument: PlatformIOImp::read_signal(): domain_idx is out of range: at geopm/src/PlatformIO.cpp:516
```

dianarg, 2020-03-09:
This seems to also happen in other components for any exception, not just geopmread.

It seems odd that anything is trying to construct an MPIComm when we have only linked with libgeopmpolicy.

dianarg, 2020-03-09:
A workaround is to use `--disable-ompt` when debugging libgeopmpolicy.  The OMPT class calls through to Policy, which tries to construct an MPIComm.
```
terminate called after throwing an instance of 'geopm::Exception'
  what():  <geopm> Invalid argument: PluginFactory::make_plugin(): name: "MPIComm" has not been previously registered: at geopm/src/PluginFactory.hpp:85

Program received signal SIGABRT, Aborted.
0x00007ffff5ddb207 in raise () from /lib64/libc.so.6
Missing separate debuginfos, use: debuginfo-install elfutils-libelf-0.172-2.el7.x86_64 glibc-2.17-260.el7_6.5.x86_64 libgcc-4.8.5-36.el7_6.2.x86_64 libstdc++-4.8.5-36.el7_6.2.x86_64 sqlite-3.7.17-8.el7.x86_64 zlib-1.2.7-18.el7.x86_64
(gdb) bt
#0  0x00007ffff5ddb207 in raise () from /lib64/libc.so.6
#1  0x00007ffff5ddc8f8 in abort () from /lib64/libc.so.6
#2  0x00007ffff6ad47d5 in __gnu_cxx::__verbose_terminate_handler() () from /lib64/libstdc++.so.6
#3  0x00007ffff6ad2746 in ?? () from /lib64/libstdc++.so.6
#4  0x00007ffff6ad2773 in std::terminate() () from /lib64/libstdc++.so.6
#5  0x00007ffff6ad2993 in __cxa_throw () from /lib64/libstdc++.so.6
#6  0x00007ffff7941941 in geopm::PluginFactory<geopm::Comm>::make_plugin (this=0x7ffff7dd9b60 <geopm::comm_factory()::instance>, plugin_name="MPIComm") at src/PluginFactory.hpp:85
#7  0x00007ffff7a400c5 in geopm::ProfileImp::ProfileImp() () at src/Profile.cpp:151
#8  0x00007ffff7a402d8 in geopm::ProfileImp::ProfileImp (this=0x7ffff7dd9c40 <geopm::Profile::default_profile()::instance>) at src/Profile.cpp:144
#9  0x00007ffff7959464 in geopm::DefaultProfile::DefaultProfile (this=0x7ffff7dd9c40 <geopm::Profile::default_profile()::instance>) at src/DefaultProfile.cpp:54
#10 0x00007ffff79593f9 in geopm::Profile::default_profile () at src/DefaultProfile.cpp:68
#11 0x00007ffff79f7910 in geopm::OMPTImp::OMPTImp (this=0x7ffff7dda240 <geopm::OMPT::ompt()::instance>) at src/OMPT.cpp:88
#12 0x00007ffff79f7879 in geopm::OMPT::ompt () at src/OMPT.cpp:82
#13 0x00007ffff7aa25f8 in ompt_initialize (lookup=0x7ffff646ea90 <_INTERNAL_26_______src_ompt_general_cpp_250e4efa::ompt_fn_lookup(char const*)>, initial_device_num=-10,
    tool_data=0x7ffff7dd96a8 <ompt_start_tool_result.6327+16>) at src/ompt_callback.cpp:64
```

It is not clear to me why ompt_initialize() is getting called for programs with no omp pragmas.  In any case, the Profile should probably be in the MPI-only section of the build if it requires a Comm.

dianarg, 2020-03-11:
It seems the comm is only used for the GEOPM_OVERHEAD part of Profile.  Could the aggregation step be moved out of Profile, and have some other MPI-dependent piece of GEOPM (like the Controller) do the reduce of the Profile overhead data?

PR #989 (2020-03-17): Move Profile and OMPT classes into libgeopm
- Profile has a dependency on Comm, and the only Comm that can be
  constructed is MPIComm.
- Fixes #980
commit e5f8ba90 (2020-03-17): Move Profile and OMPT classes into libgeopm

- Profile has a dependency on Comm, and the only Comm that can be
  constructed is MPIComm.
- Fixes #980

Change-Id: I6896fe987d2ef8cf7e44816559b6ff26e9436c0e
closed 2020-03-17 by dianarg