geopm/geopm #1047: Updating missing documentation for Helper.hpp

sidjana, 2020-04-15:


Rishabh-Maheshwari, 2021-11-12:
I would like to work on this issue.

dianarg, 2021-11-12:
@Rishabh-Maheshwari Go ahead!  Let us know if you need any help.

Rishabh-Maheshwari, 2021-11-12:
Thank You, @dianarg for the Go Ahead! Could you please provide description for the following functions:

1. [read_double_from_file()](https://github.com/geopm/geopm/blob/2a1854e73c7ae8adea263dfec75ac7466361a4aa/service/src/Helper.cpp#L79)
1. [hostname()](https://github.com/geopm/geopm/blob/2a1854e73c7ae8adea263dfec75ac7466361a4aa/service/src/Helper.cpp#L151)
1. [check_hint()](https://github.com/geopm/geopm/blob/2a1854e73c7ae8adea263dfec75ac7466361a4aa/service/src/Helper.cpp#L237)
1. [string_format_type_to_function()](https://github.com/geopm/geopm/blob/2a1854e73c7ae8adea263dfec75ac7466361a4aa/service/src/Helper.cpp#L251)
1. [string_format_function_to_type()](https://github.com/geopm/geopm/blob/2a1854e73c7ae8adea263dfec75ac7466361a4aa/service/src/Helper.cpp#L267)
1. [get_env()](https://github.com/geopm/geopm/blob/2a1854e73c7ae8adea263dfec75ac7466361a4aa/service/src/Helper.cpp#L284)

Also, Can you please **assign** the issue to me?

dianarg, 2021-11-16:
@Rishabh-Maheshwari Here is how I would describe those functions:

- read_double_from_file(): reads a double value from the file located at _file_ and check that the units, if present, match _expected_units_.
    
- hostname(): returns the hostname of the local host.  [you might want to add a link to https://www.man7.org/linux/man-pages/man2/gethostname.2.html for this one]
    
- check_hint(): checks that the given integer _hint_ contains only a single non-zero hint bit in the valid range for hints.
    
- string_format_type_to_function(): returns the function corresponding to the given integer _format_type_.
    
- string_format_function_to_type(): returns the corresponding enum value for a given string _format_function_.  
    
- get_env(): returns the value of the environment variable _name_ as a C++ std::string.

In the ronn descriptions, please make sure all parameters for each function are mentioned by name; I tried to put them in italics above.