geopm/geopm #1083: GEOPM gtests fail with version of ICC in parallel studio xe cluster edition 2019 update 5

eastep, 2020-04-30:
All gtests fail with this version of parallel studio on the team's research system containing GPUs. Updating to a 2020 version of parallel studio fixes the errors. This release of parallel studio contains a newer version of the intel compiler toolchain. Perhaps that's why. The error logs for the gtests suggest that the old version of the toolchain did not support new enough libc ABIs. This may be a problem when running GEOPM with CUDA programs on such a system because the most recent release of CUDA does not support the newer version of the intel compiler toolchain.

eastep, 2020-04-30:
There hasn't been a release of CUDA that I'm aware of since November 28, 2019 so perhaps a new one will be available soon. Perhaps before ISC'20? My contacts either don't know or won't say.

cmcantalupo, 2020-04-30:
@eastep Can you paste the error messages into the bug please?

eastep, 2020-04-30:
@cmcantalupo  yes, of course -- once the system is freed up again.

eastep, 2020-05-04:
% cat test/gtest_links/AdminTest.agent_no_policy.log
_./test/gtest_links/../.libs/geopm_test: /opt/intel/inspector_2019.5.0.602103/lib64/libstdc++.so.6: version `CXXABI_1.3.11' not found (required by ./test/gtest_links/../.libs/geopm_test)_
_./test/gtest_links/../.libs/geopm_test: /opt/intel/inspector_2019.5.0.602103/lib64/libstdc++.so.6: version `GLIBCXX_3.4.22' not found (required by ./test/gtest_links/../.libs/geopm_test)_

eastep, 2020-05-04:
Further investigation shows that there is actually a version of libstdc++.so available on the system (from Intel Parallel Studio XE Cluster Edition 2019 Update 5) which provides new enough ABIs that the gtests work. I tried it, and it worked. This version of parallel studio has compilers that are compatible with the version of CUDA installed on the system. So, the original problem I raised has a straightforward solution: switch to using Parallel Studio XE Cluster Edition 2019 Update 5. 

However, this whole exercise makes me wonder if GEOPM has a dependency on a version of libstd++.so that is so new that some sites might not have it installed. The version in inspector_2019.5.0.602103 can't have been much older than 2019. Do we fundamentally need such a recent version of libstd++ for our tests (or other features) to work, or is there a way to avoid that?

cmcantalupo, 2020-05-11:
It appears from the error message that you built geopm/test/geopm_test.c with a different toolchain than the one that you had in your LD_LIBRARY_PATH at runtime.  If you change your compiler toolchain, you must clean out all of the existing object files, reconfigure and recompile.  I'm going to close this issue.  Please let me know if the procedure below solves your problem.  

```
$ make clean
$ make distclean
$ ./autogen.sh
$ ./configure <WITH_NEW_ARGUMENTS>
$ make
```

closed 2020-05-11 by cmcantalupo

eastep, 2020-05-12:
@cmcantalupo  I tried that previously (and again just now). Doesn't fix the problem. The compiler toolchain wasn't changed between building GEOPM and running it. In any case, I found a way around this issue by simply using a different version of the toolchain (as noted above), so I have no objection to closing this issue and moving on.