geopm/geopm #1124: Write unit tests and implementation for RegionProcessEpoch

cmcantalupo, 2020-05-27:
> estimate 

The RegionProcessEpoch will infer the epoch signals based on the user specifying the parameters below in the environment:

GEOPM_REGION_EPOCH=MPI_Barrier
GEOPM_REGION_EPOCH_OFFSET=10
GEOPM_REGION_EPOCH_STRIDE=2

This will enable the MPI_Barrier region to be interpreted as an epoch for every two occurrences after the first 10 occurrences of the region.  This will be a very simple epoch inference algorithm that can be a base line for more elaborate epoch signal inference.  The pseudo code below is an example where the parameters above is useful.

```
// Start up code
for (i : range(10)) {
    MPI_Barrier();
    do_stuff();
}
// Main loop with an epoch
for (i : range(10000) {
    /// Inferred geopm_prof_epoch() call
    MPI_Barrier();
    do_loop_stuff1();
    MPI_Barrier();
    do_loop_stuff2();
}
```

Done criteria:
Implementation and unit tests merged

cmcantalupo, 2020-06-02:
related to #1137 

We have divided the responsibilities listed in this issue into different places.  Now the ProcessEpochImp is not responsible for doing epoch inference.  This is done with a RecordFilter (e.g. the ProxyEpochRecordFilter).  The selection of the filter and the parameters for the filter should be the responsibility of the EpochIOGroup #1125.  We will also have to add these parameters to the Environment class.

dianarg, 2020-06-05:
Implemented as ProxyEpochRecordFilter in #1139

closed 2020-06-05 by dianarg