geopm/geopm #1178: Manually run miniFE, nasFFT, nekbone, and gadget with existing script, but using edit distance filter

cmcantalupo, 2020-06-24:
> estimate

Modify existing scripts used to generate test profile trace data to actually run with the edit distance filter.  Remove all geopm modifications from benchmarks.  Run with the monitor.

Done criteria:
- The runtime doesn't fail
- Examine profile trace and EPOCH_COUNT events

Parent story: #1173

fkeceli, 2020-06-30:
Few questions/comments:

- Is this referring to adding all these apps to GEOPM as an integration test or is it just a step towards it?
- Note that GADGET is not fully detected by APD unless we do some manual trace filtering which is not implemented in GEOPM.
- Ali is working on adding HACC to this.

dianarg, 2020-07-06:
Nekbone, minife, and FFT can run successfully.  Issue opened for gadget: #1188

closed 2020-07-06 by dianarg