geopm/geopm #1193: As a user of the Gadget application, I would like to understand the opportunities that exist for saving energy with GEOPM

cmcantalupo, 2020-07-08:


bgeltz, 2020-08-04:
http://kronos.ra.intel.com/~brad/data/smng/gadget/766273_gadget/region_mean_stats.log
Generated from:
https://github.com/bgeltz/dotfiles/blob/smng/bin/bin/gen_mean_stats.py

bgeltz, 2020-08-10:
I just wrote some new code to print the per-iteration times of all the regions.  It's here:
https://github.com/bgeltz/dotfiles/blob/smng/bin/bin/gen_region_length.py

It will generate a report that is filtered by these 2 criteria:
1. The per-iteration time is > 5ms.
2. The total runtime is > 10s.

The filtered report is here:
http://kronos.ra.intel.com/~brad/data/smng/gadget/766273_gadget/region_iteration_runtime_filtered.log
(Unfiltered: http://kronos.ra.intel.com/~brad/data/smng/gadget/766273_gadget/region_iteration_runtime.log)

Regarding the energy savings opportunity from the done criteria:
There's a mix of long running and short running regions.  The list of non-MPI regions that ran for > 100 ms is:
- [OMPT]gravity_tree+0x595
- [OMPT]hydro_force+0x295

While the two above regions comprise > 50% of the run, both regions are frequency sensitive.  No other regions have a significant enough per-iteration or total runtime.

There are a few long running collective calls with per-iteration times > 10ms:
- MPI_Alltoall
- MPI_Waitall

MPI_Allreduce did have a large runtime, but the per-iteration times are ~=5ms.

closed 2020-08-10 by bgeltz