geopm/geopm #1266: Build failure in absence of ronn despite --disable-ronn flag in configuration

rountree, 2020-08-26:
Short version:  ronn isn't installed on Quartz.  Standard configuration doesn't catch this and the build eventually fails.  Configuring with --disable-ronn doesn't turn off all uses of ronn, and the build still fails.

Two issues here:  catch the absence of ronn in the configuration phase, and have --disable-ronn work across the entire project.

Long version:

Building from commit 771aae78....  (Fri Aug 14 2020).  Running on quartz.llnl.gov.

Configuring as:

```
../../repos/geopm/configure --prefix=${HOME}/v/install/geopm
```
results in 
```
version            : 1.1.0+dev463g771aae7
...
debug              : 0
coverage           : 0
overhead           : 0
mpi                : 1
openmp             : 1
fortran            : 1
ronn               : 1
ompt               : 0
beta               : 0
bloat              : 0
nvml               : 0
```

But the subsequent build fails on missing ronn.
```
mkdir -p man
export GEM_PATH=`readlink -f ../../repos/geopm/ruby`:$GEM_PATH && \
cd ronn && \
../ruby/bin/ronn --date=`date "+%Y-%m-%d"` \
                 --organization="GEOPM 1.1.0+dev463g771aae7" \
                 `basename man/GEOPM_CXX_MAN_Agent.3`.ronn
/usr/share/rubygems/rubygems/dependency.rb:296:in `to_specs': Could not find 'ronn' (>= 0) among 0 total gem(s) (Gem::LoadError)
        from /usr/share/rubygems/rubygems/dependency.rb:307:in `to_spec'
        from /usr/share/rubygems/rubygems/core_ext/kernel_gem.rb:47:in `gem'
        from ../ruby/bin/ronn:22:in `<main>'
make[1]: *** [man/GEOPM_CXX_MAN_Agent.3] Error 1
make[1]: Leaving directory `/p/vast1/rountree/build/geopm'
make: *** [all] Error 2
```

After configuring with `--disable-ronn` the build fails as follows:
```
test -f man/GEOPM_CXX_MAN_EnergyEfficientRegion.3 || sed 's|^\[//\]: #|.\\\"|' ronn/`basename man/GEOPM_CXX_MAN_EnergyEfficientRegion.3`.ronn > man/GEOPM_CXX_MAN_EnergyEfficientRegion.3
sed: can't read ronn/GEOPM_CXX_MAN_EnergyEfficientRegion.3.ronn: No such file or directory
make[1]: *** [man/GEOPM_CXX_MAN_EnergyEfficientRegion.3] Error 2
make[1]: Leaving directory `/p/vast1/rountree/build/geopm'
make: *** [all] Error 2
```

cmcantalupo, 2020-08-26:
@rountree You are building out of tree.  The error you are seeing looks like a duplicate of of you're previously reported error:

https://github.com/geopm/geopm/issues/820

@bakerbrandond is working on a fix:

https://github.com/geopm/geopm/pull/1257

cmcantalupo, 2020-08-26:
@rountree #1257 was just merged to the dev branch.  Please let me know if this resolves your out of place build issues with the man pages (this issue and #820).  Thanks for your feedback.

rountree, 2020-08-27:
I completely forgot about that error report (although it was back in November last year).  I thought it got fixed?

Anyway.... 

```
version            : 1.1.0+dev466gd9ec167
```

fails here:

```
test -f man/GEOPM_CXX_MAN_Agent.3 || sed 's|^\[//\]: #|.\\\"|' ronn/`basename man/GEOPM_CXX_MAN_Agent.3`.ronn > man/GEOPM_CXX_MAN_Agent.3
sed: can't read ronn/GEOPM_CXX_MAN_Agent.3.ronn: No such file or directory
make[1]: *** [man/GEOPM_CXX_MAN_Agent.3] Error 2
make[1]: Leaving directory `/p/vast1/rountree/build/geopm'
make: *** [all] Error 2
```

The `ronn/GEOPM_CXX_MAN_Agent.3.ronn` exists in the source tree, of course.  And manually copying all of the files from that source directory to the build directory allowed the build to complete without error.

Barry

bakerbrandond, 2020-08-27:
> I completely forgot about that error report (although it was back in November last year). I thought it got fixed?
>
> Anyway....
>
> ```
> version            : 1.1.0+dev466gd9ec167
> ```
>
> fails here:
>
> ``` [...]

Hey Barry,

We had the patch to fix this up merged yesterday, I don't recognize the commit hash listed in this issue but please make sure you pull the latest, need to have 6f8c9ed9757fe4af3695e2fe526489abb965c92e at least but dev is currently at d9ec16723d8db5a71a0c322ed3a0f754a5138ab7

bakerbrandond, 2020-08-27:
> I completely forgot about that error report (although it was back in November last year). I thought it got fixed?
>
> Anyway....
>
> ```
> version            : 1.1.0+dev466gd9ec167
> ```
>
> fails here:
>
> ``` [...]

Ok it seems that your hash does checkout.  Have you tried completely nuking your existing build dir?  If the repro steps have changed from your original issue please provide updated steps.

After checkout out geopm and cd'ing into the source dir, running ```autogen.sh from there, mkdir build; cd build; ../configure;```

I am able to build and install with the current tip of dev.  Is there something different in your setup?

rountree, 2020-08-27:
Hmmm..... 

```
rountree@quartz764 ~/v/repos/geopm$ date
Thu Aug 27 11:30:42 PDT 2020
rountree@quartz764 ~/v/repos/geopm$ git pull upstream dev
X11 forwarding request failed
From github.com:geopm/geopm
 * branch            dev        -> FETCH_HEAD
Already up-to-date.
rountree@quartz764 ~/v/repos/geopm$ git rev-parse HEAD
d9ec16723d8db5a71a0c322ed3a0f754a5138ab7
rountree@quartz764 ~/v/repos/geopm$ ./autogen.sh 
libtoolize: putting auxiliary files in `.'.
libtoolize: copying file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: copying file `m4/libtool.m4'
libtoolize: copying file `m4/ltoptions.m4'
libtoolize: copying file `m4/ltsugar.m4'
libtoolize: copying file `m4/ltversion.m4'
libtoolize: copying file `m4/lt~obsolete.m4'
rountree@quartz764 ~/v/repos/geopm$ cd ../../buil
build/        build-lassen/ 
rountree@quartz764 ~/v/repos/geopm$ cd ../../build/geopm/
rountree@quartz764 ~/v/build/geopm$ ls
myconfigure.sh  mydistclean.sh
rountree@quartz764 ~/v/build/geopm$ cat ./myconfigure.sh 
#!/bin/bash

../../repos/geopm/configure --disable-ronn --prefix=${HOME}/v/install/geopm
```

Parsing the output of that....

```
version            : 1.1.0+dev466gd9ec167
```

The build still fails:
```
1584 libtool: link: gfortran -fopenmp -msse4.2 -I/usr/tce/packages/mvapich2/mvapich2-2.3-gcc-8.1.0/include -I/usr/tce/packages/mvapich2/mvapich2-2.3-gcc-8.1.0/include -O3 -fopenmp -Wl,-rpath -Wl,/usr/tc     e/packages/mvapich2/mvapich2-2.3-gcc-8.1.0/lib -o examples/fft/.libs/nas_ft examples/fft/examples_fft_nas_ft-ft.o examples/fft/examples_fft_nas_ft-print_results.o examples/fft/examples_fft_nas_ft-r     andi8.o examples/fft/examples_fft_nas_ft-timers.o  -L/usr/tce/packages/mvapich2/mvapich2-2.3-gcc-8.1.0/lib ./.libs/libgeopm.so -L/builddir/build/BUILD/mvapich2-2.3/src/mpl ./.libs/libgeopmfort.so /     g/g24/rountree/v/build/geopm/.libs/libgeopm.so /usr/tce/packages/gcc/gcc-8.1.0/lib64/../lib64/libgomp.so /usr/tce/packages/mvapich2/mvapich2-2.3-gcc-8.1.0/lib/libmpifort.so /usr/tce/packages/gcc/gc     c-8.1.0/lib64/../lib64/libgfortran.so -lz /usr/tce/packages/gcc/gcc-8.1.0/lib64/../lib64/libquadmath.so /usr/tce/packages/mvapich2/mvapich2-2.3-gcc-8.1.0/lib/libmpicxx.so /usr/tce/packages/gcc/gcc-     8.1.0/lib64/../lib64/libstdc++.so /usr/tce/packages/mvapich2/mvapich2-2.3-gcc-8.1.0/lib/libmpi.so -lm -lnuma -libverbs -lpsm2 -lpmi2 -lelf -lpthread -ldl -lrt -fopenmp -Wl,-rpath -Wl,/g/g24/rountre     e/v/install/geopm/lib -Wl,-rpath -Wl,/usr/tce/packages/gcc/gcc-8.1.0/lib64/../lib64 -Wl,-rpath -Wl,/usr/tce/packages/mvapich2/mvapich2-2.3-gcc-8.1.0/lib
1585 mkdir -p man
1586 test -f man/GEOPM_CXX_MAN_Agent.3 || sed 's|^\[//\]: #|.\\\"|' ronn/`basename man/GEOPM_CXX_MAN_Agent.3`.ronn > man/GEOPM_CXX_MAN_Agent.3
1587 sed: can't read ronn/GEOPM_CXX_MAN_Agent.3.ronn: No such file or directory
1588 make[1]: *** [man/GEOPM_CXX_MAN_Agent.3] Error 2
1589 make[1]: Leaving directory `/p/vast1/rountree/build/geopm'
1590 make: *** [all] Error 2
```

This might help:
```
rountree@quartz764 ~/v/build/geopm$ ls ronn
geopm.7.ronn
```

Can you point me at the code that should be copying the other ronn files into the build directory?  I'm guessing this is a software version problem.

Thanks!

bakerbrandond, 2020-08-27:
> Hmmm.....
>
> ```
> rountree@quartz764 ~/v/repos/geopm$ date
> Thu Aug 27 11:30:42 PDT 2020
> rountree@quartz764 ~/v/repos/geopm$ git pull upstream dev
> X11 forwarding request failed
> From [...]

The code of interest is here: https://github.com/geopm/geopm/blob/dev/Makefile.am#L831-L849

I am able to make and install with a build dir that is a subdir or geopm src, or completely out of the geopm src dir.  I am successful with full paths in configure as well as relative.  I haven't logged into quartz in a while, maybe I can try to repro in your environment.

bakerbrandond, 2020-08-27:
I am working to resolve something I noticed in the log since you mentioned that this issue still persists
```if [ $TARGET_DOC != "geopm.7" ] ; then \
        TMP_SRC="/home/bbaker1/dev/geopm/ronn/$TARGET_DOC" ; \
else \
        TMP_SRC="$RONN_DIR/$TARGET_DOC" ; \
fi ; \
cat "./ronn/header.txt" "$TMP_SRC.ronn" | \
        sed -e 's/\(#include.*\) \\fIhttp.*/\1\\fR/g' > man/geopmlaunch.1
/bin/sh: line 0: [: !=: unary operator expected
cat: /.ronn: No such file or directory```

Working on a fix, maybe causing your issue?  Not sure why we are seeing differences in build result error codes.

rountree, 2020-08-27:
That is a really cool thing you did there with the github lines-of-code link.  But....  shouldn't that stanza be disabled if configured with the `--disable-ronn` flag passed to the configure script (as I did above)?

bakerbrandond, 2020-08-27:
> That is a really cool thing you did there with the github lines-of-code link. But.... shouldn't that stanza be disabled if configured with the `--disable-ronn` flag passed to the configure script (as I did above)?

You are correct.  I am conflating 2 issues, first the issue with the out of source configuration build and failure to make and install due to ronn dir path differences (still needs work); the second issue [issue 1266] if --disable-ronn is presented this should all be a no-op.

PR #1312 (2020-09-22): Clean up man page generation for out of tree builds
- Move ronn logic into bash script.
- Don't use configure to insert BLURB, do it in the bash script.
- Get rid of --disable-ronn configure option (copy md with warning)
- Fixes #1266
closed 2020-09-22 by bgeltz