geopm/geopm #1321: Replace logic error blocks with GEOPM_DEBUG_ASSERT macro

dianarg, 2020-09-27:
Within the GEOPM project, a logic error is a type of error that should be _impossible_ to reach given the logic of the code.  For example, they can be used in situations where the caller is guaranteed not to pass a null pointer, so receiving a null pointer as input indicates a problem with the calling code.  In the past, these exceptions have been implemented like the following:
```
void func(int input) {
#ifdef GEOPM_DEBUG
    if (input < 0) {
        throw Exception("Input passed was negative", GEOPM_ERROR_LOGIC, __FILE__, __LINE__);
    }
#endif
}
```
`GEOPM_DEBUG` is set at configure time by `--enable-debug`.  In release builds, code in these blocks is removed.

The `GEOPM_DEBUG_ASSERT(condition, fail_message)` macro can be used to implement the same pattern with less repeated code.  To use it,  geopm_debug.hpp must be included.  The code above could be reimplemented as the following:
```
void func(int input) {
    GEOPM_DEBUG_ASSERT(input >=0, "Input passed was negative");
}
```
**Note**: the condition passed to GEOPM_DEBUG_ASSERT will be the opposite of the condition from the if-statement.  This interface is intended to be similar to `assert()` from https://en.cppreference.com/w/cpp/error/assert

Rishabh-Maheshwari, 2021-11-12:
I would like to work on this.

dianarg, 2021-11-12:
@Rishabh-Maheshwari  Go for it!  Let us know if you need any help.  For this type of issue, it's perfectly fine to submit multiple pull requests for different files.