geopm/geopm #1475: ProfileIOGroup reports available signals without application connection

dianarg, 2021-01-06:
This issue is for the new-profile branch.

`geopmread` lists various PROFILE:: signals as available, but they are not valid; trying to read gives the error:
```
> ./geopmread PROFILE::REGION_HASH board 0
Error: cannot read signal: <geopm> Runtime error: ApplicationSamplerImp::cpu_region_hash(): cannot read process info before connect().: at geopm/src/ApplicationSampler.cpp:265
```
These signals should either return NAN when no application is connected, or not be present in the list of available signals.

dianarg, 2021-01-22:
This is inconsistent with how the MSRIOGroup works.  It might be nice to be able to use `geopmread --info` for some of these signals, even if they can't be read (e.g. because the msr-safe driver is not present).

cmcantalupo, 2022-03-24:
The way this should be handled by each IOGroup should be consistent and documented by 2.0.

cmcantalupo, 2022-07-20:
This is no longer an issue.  ProfileIOGroup is only loaded when the HPC Runtime singleton accessor PlatformIOProf::platform_io() is used to get access to PlatformIO.

closed 2022-07-20 by cmcantalupo