geopm/geopm #1483: FrequencyMapAgent report extensions not proper YAML

dianarg, 2021-01-11:
The FrequencyMapAgent adds some information to the host report about the frequency map that cannot be parsed as YAML.

Related to #615

bgeltz, 2022-03-28:
Example report:
```
GEOPM Version: 1.1.0+dev1927g3ecc408c
Start Time: Mon Mar 28 14:06:03 2022
Profile: test_frequency_map
Agent: frequency_map
Policy: {"FREQ_DEFAULT": 2100000000, "FREQ_UNCORE": "NAN", "HASH_0": 2806759221, "FREQ_0": 1200000000, "HASH_1": 3599882752, "FREQ_1": 1900000000, "HASH_2": 1037861311, "FREQ_2": 1000000000, "HASH_3": "NAN", "FREQ_3": "NAN", "HASH_4": "NAN", "FREQ_4": "NAN", "HASH_5": "NAN", "FREQ_5": "NAN", "HASH_6": "NAN", "FREQ_6": "NAN", "HASH_7": "NAN", "FREQ_7": "NAN", "HASH_8": "NAN", "FREQ_8": "NAN", "HASH_9": "NAN", "FREQ_9": "NAN", "HASH_10": "NAN", "FREQ_10": "NAN", "HASH_11": "NAN", "FREQ_11": "NAN", "HASH_12": "NAN", "FREQ_12": "NAN", "HASH_13": "NAN", "FREQ_13": "NAN", "HASH_14": "NAN", "FREQ_14": "NAN", "HASH_15": "NAN", "FREQ_15": "NAN", "HASH_16": "NAN", "FREQ_16": "NAN", "HASH_17": "NAN", "FREQ_17": "NAN", "HASH_18": "NAN", "FREQ_18": "NAN", "HASH_19": "NAN", "FREQ_19": "NAN", "HASH_20": "NAN", "FREQ_20": "NAN", "HASH_21": "NAN", "FREQ_21": "NAN", "HASH_22": "NAN", "FREQ_22": "NAN", "HASH_23": "NAN", "FREQ_23": "NAN", "HASH_24": "NAN", "FREQ_24": "NAN", "HASH_25": "NAN", "FREQ_25": "NAN", "HASH_26": "NAN", "FREQ_26": "NAN", "HASH_27": "NAN", "FREQ_27": "NAN", "HASH_28": "NAN", "FREQ_28": "NAN", "HASH_29": "NAN", "FREQ_29": "NAN", "HASH_30": "NAN", "FREQ_30": "NAN"}

Hosts:
  mcfly1:
    Frequency map:
    0x000000003ddc81bf: 1.0000000000000000e+09
    0x00000000a74bbf35: 1.2000000000000000e+09
    0x00000000d691da00: 1.9000000000000000e+09
    0x00000000644f9787: 2.1000000000000000e+09
    0x00000000725e8066: 2.1000000000000000e+09
    0x000000007b561f45: 2.1000000000000000e+09
    0x8000000000000000: 2.1000000000000000e+09

    Regions:
    -
      region: "all2all"
      hash: 0x3ddc81bf
      runtime (s): 5.04204
      count: 5
      sync-runtime (s): 5.04142
      package-energy (J): 659.813
      dram-energy (J): 73.1866
      power (W): 130.878
      frequency (%): 51.7619
      frequency (Hz): 1.087e+09
      time-hint-network (s): 5.04063
      time-hint-ignore (s): 0
      time-hint-compute (s): 0
      time-hint-memory (s): 0
      time-hint-io (s): 0
      time-hint-serial (s): 0
      time-hint-parallel (s): 0
      time-hint-unknown (s): 0.000700021
      time-hint-unset (s): 0
      frequency-map: 1000000000.000000
    -
      region: "model-init"
      hash: 0x644f9787
      runtime (s): 3.59365
      count: 1
      sync-runtime (s): 3.55971
      package-energy (J): 913.291
      dram-energy (J): 82.6084
      power (W): 256.563
      frequency (%): 97.9037
      frequency (Hz): 2.05598e+09
      time-hint-network (s): 0
      time-hint-ignore (s): 0.00349992
      time-hint-compute (s): 0.182506
      time-hint-memory (s): 0
      time-hint-io (s): 0
      time-hint-serial (s): 0
      time-hint-parallel (s): 0
      time-hint-unknown (s): 3.37366
      time-hint-unset (s): 0
      frequency-map: 2100000000.000000
    -
      region: "dgemm"
      hash: 0xa74bbf35
      runtime (s): 3.32094
      count: 5
      sync-runtime (s): 3.30628
      package-energy (J): 718.383
      dram-energy (J): 56.5331
[... 111 lines cut ...]
      geopmctl network BW (B/s): 0
```

The problematic section is:
```
    Frequency map:
    0x000000003ddc81bf: 1.0000000000000000e+09
    0x00000000a74bbf35: 1.2000000000000000e+09
    0x00000000d691da00: 1.9000000000000000e+09
    0x00000000644f9787: 2.1000000000000000e+09
    0x00000000725e8066: 2.1000000000000000e+09
    0x000000007b561f45: 2.1000000000000000e+09
    0x8000000000000000: 2.1000000000000000e+09
```

Which manifests like this when you try to use the `RawReport`:
```
>>> self._report.raw_report()['Hosts']['mcfly1']
{'Frequency map': None, 1037861311: 1000000000.0, 2806759221: 1200000000.0, 3599882752: 1900000000.0, 1682937735: 2100000000.0, 1918795878: 2100000000.0, 2069241669: 2100000000.0, 9223372036854775808: 2100000000.0, 'Regions': [{'region': 'all2all', 'hash': 1037861311,...
```
So if you tried to print the map, you get `None` back:
```
>>> self._report.raw_report()['Hosts']['mcfly1']['Frequency map']
>>>
```

dannosliwcd, 2022-03-30:
Consider formatting/quoting custom report outputs with json11 since we already have that as a dependency, and it should (right?) also result in something that can be correctly interpreted as yaml.

Also consider whether general fields could result in yaml problems. E.g., `--geopm-profile=[1,2,3]` or weirdly-named apps or something similar.

PR #2255 (2022-04-25): Changes to FrequencyMapAgent to solve the JSON bug
The FrequencyMapAgent adds some information to the host report about the frequency map that cannot be parsed as YAML.

This fixes the frequency map so that it becomes valid JSON, which can then be parsed by the python interpreter.

- Relates to #1483 
- Fixes #1483
closed 2022-04-25 by bgeltz