geopm/geopm #1517: Invalid report when application has no regions

dianarg, 2021-01-21:
When an application has no regions (user marked, MPI, or otherwise) the `Regions:` section of the report contains nothing.  It should be an empty list, e.g. `Regions: [] `.  The report parsing and caching in io.py also needs to be updated to support an empty regions dataframe.

A workaround is to add at least one MPI call to any application with no regions.

See #1513

cmcantalupo, 2022-03-23:
Related to
https://github.com/geopm/geopm/issues/1570
?

bgeltz, 2022-03-28:
This was done already:
https://github.com/geopm/geopm/issues/1570#issuecomment-1081137292

Thought we still need to update io.py:
```
======================================================================
ERROR: test_count (__main__.TestIntegration_monitor)
Test that region and epoch counts in the report match expected
----------------------------------------------------------------------
Traceback (most recent call last):
  File "./test_monitor.py", line 106, in test_count
    spin_data = self._report.raw_region(node, 'spin')
  File "/home/bgeltz/build/geopm/lib/python3.6/site-packages/geopmpy/io.py", line 785, in raw_region
    for rr in self._raw_dict['Hosts'][host_name]['Regions']:
KeyError: 'Regions'
```

PR #2478 (2022-07-27): Handle report parsing with no user defined or OMPT regions
- Fixes #1517.
commit d5dc86a7 (2022-07-27): Handle reports with no user defined or OMPT regions

- Fixes #1517.
PR #2529 (2022-08-09): Fixup RawReportCollection handling of reports with no regions
- Fixes #1517.
closed 2022-08-09 by cmcantalupo