geopm/geopm #1532: NVML Frequency controls required root permission

lhlawson, 2021-02-03:
The NVML API call used for NVML::FREQUENCY_CONTROL requires admin level permissions, limiting the number of users that can set frequency.

Adding NVML::APPLICATION_FREQUENCY_CONTROL to use nvmlDeviceSetApplicationsClocks will provide additional non-root users control on systems that are set up with 
nvidia-smi -acp 0

cmcantalupo, 2022-03-23:
The GEOPM service may be used to provide access, which is a different way to deal with this problem.   I the nvidia-smi -acp 0 configuration common?  If so, perhaps the suggested solution would be preferred to requiring the geopm-service to be active.

lhlawson, 2022-03-23:
nvidia-smi -acp 0 is not the default behavior, but is well documented by NVIDIA (since [2014](https://developer.nvidia.com/blog/increase-performance-gpu-boost-k80-autoboost/)) as a way for non-root users to be allowed to control frequency.  

That said, current [AWS documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/optimize_gpu.html) still lists using sudo to set frequency on GPU enabled platforms. 

It's a relatively small change (patch [here](https://github.com/lhlawson/geopm/commit/97d096d68219dc5f493ea58b3c40dc00b27eb3be)) to support application clocks as well as the current locked clocks if required.