geopm/geopm #1570: When application has no regions, the region sectionis still included in report

cmcantalupo, 2021-03-08:
Diana started work on this branch here:

https://github.com/dianarg/geopm/tree/dianarg-new-profile-report

cmcantalupo, 2022-03-23:
Maybe this is fixed now?

bgeltz, 2022-03-28:
This has been done already:
```
GEOPM Version: 1.1.0+dev1927g3ecc408c
Start Time: Mon Mar 28 13:56:05 2022
Profile: test_monitor
Agent: monitor
Policy: {}

Hosts:
  mcfly1:
    Unmarked Totals:
      runtime (s): 20.0057
      count: 0
      sync-runtime (s): 20.0057
      package-energy (J): 2215.74
      dram-energy (J): 211.169
      power (W): 110.755
      frequency (%): 119.353
      frequency (Hz): 2.50642e+09
      time-hint-network (s): 0
      time-hint-ignore (s): 0
      time-hint-compute (s): 0
      time-hint-memory (s): 0
      time-hint-io (s): 0
      time-hint-serial (s): 0
      time-hint-parallel (s): 0
      time-hint-unknown (s): 0
      time-hint-unset (s): 20.0052
    Epoch Totals:
      runtime (s): 20.0006
      count: 20
      sync-runtime (s): 20.0006
      package-energy (J): 2215.07
      dram-energy (J): 211.115
      power (W): 110.75
      frequency (%): 119.344
      frequency (Hz): 2.50622e+09
      time-hint-network (s): 0
      time-hint-ignore (s): 0
      time-hint-compute (s): 0
      time-hint-memory (s): 0
      time-hint-io (s): 0
      time-hint-serial (s): 0
      time-hint-parallel (s): 0
      time-hint-unknown (s): 0
      time-hint-unset (s): 20.0002
    Application Totals:
      runtime (s): 20.0057
      count: 0
      sync-runtime (s): 20.0057
      package-energy (J): 2215.74
      dram-energy (J): 211.169
      power (W): 110.755
      frequency (%): 119.353
      frequency (Hz): 2.50642e+09
      time-hint-network (s): 0
      time-hint-ignore (s): 0
      time-hint-compute (s): 0
      time-hint-memory (s): 0
      time-hint-io (s): 0
      time-hint-serial (s): 0
      time-hint-parallel (s): 0
      time-hint-unknown (s): 0
      time-hint-unset (s): 20.0052
      geopmctl memory HWM (B): 112236
      geopmctl network BW (B/s): 0
```

closed 2022-03-28 by bgeltz