geopm/geopm #1671: As a user of PlatformIO's high level aliases I would like a more explicit naming scheme

cmcantalupo, 2021-06-16:
The high level alias names for signals and controls should be vetted and renamed since these strings play critical role in the service interface, and some names are too general (e.g. FREQUENCY).

There is also an ambiguity about what it means to read a control, this could be resolved by appending the string "control" to all controls and then reading this value would use the same string in an unambiguous way.  This would, for instance, disambiguate the requested frequency from the obtained frequency.

dannosliwcd, 2022-03-16:
Related: #2169 requests signal aliases for power control MSRs. But the existing control aliases do not follow the new recommended naming scheme.

cmcantalupo, 2022-03-16:
Suggestion is that the naming scheme be of the following form:

```
[<COMPONENT>]_NAME_[CONTROL/STATUS]
```

`COMPONENT`
---------------
This is one of the following strings: "CPU", "DRAM", "GPU", or "NIC".  This should always be provided in the alias if the signal or control pertains to one of these components.  An alias like "TIME" would not have this prefix.  

Note that the native domain of a signal or control may not match the component name.  For example, the RAPL energy counter is native to the package domain, and is currently aliased to "ENERGY_PACKAGE", but after the rename this will become "CPU_ENERGY".  The user will have to query the PlatformIO::signal_domain() or PlatformIO::control_domain() method to determine the native domain rather than having that reflected in the name.  

`CONTROL` or `STATUS`
--------------------------
The string "CONTROL" or "STATUS" may be provided to disambiguate a setting from the current state.  All control alias strings will end with "CONTROL".  All control aliases will have a corresponding signal with the same name which enables the user to read the current setting for a control value.  THE "STATUS" string is used to specify the current state of the system as it is measured.  For example:

- "CPU_FREQUENCY_MAX_CONTROL" control alias enables the setting of the maximum core frequency
- "CPU_FREQUENCY_MAX_CONTROL" signal alias will reflect the current setting for this control
- "CPU_FREQUENCY_MAX" signal alias will reflect the maximum value that the control may be set to
- "CPU_FREQUENCY_STATUS" signal will reflect the operating frequency of a core.

bgeltz, 2022-03-17:
Trying to disambiguate between setting the non-turbo maximum core frequency vs. turbo might be something like:
```
CPU_FREQUENCY_TURBO_MAX_CONTROL
CPU_FREQUENCY_MAX_CONTROL
CPU_FREQUENCY_MAX
CPU_FREQUENCY_STATUS
```

closed 2022-06-21 by cmcantalupo