geopm/geopm #1676: As a developer of the GEOPM service I would like to ensure that all user input is properly vetted

cmcantalupo, 2021-06-16:
It is especially important to test that the service is robust in all user facing interfaces to bad or malicious input.

cmcantalupo, 2022-03-25:
Requires review of existing tests and determine if there are existing gaps.  Many if not all of our user facing interfaces have robust unit tests.

dannosliwcd, 2022-05-10:
One known case is geopm profile names. Users can input `--geopm-profile=[0, 1, 2, anything that parses as a complex yaml data structure]` and it will bubble through to our report files which will be interpreted as yaml. In that specific case, a solution could be to write the profile (or any field) to the report in a way that ensures it is interpreted as a string, e.g., by using a full yaml-aware dumper.

closed 2022-07-19 by cmcantalupo