geopm/geopm #1680: As a user of the GEOPM reports, I would like to be able to specify the aggregation fuction applied over time to "report signals"

cmcantalupo, 2021-06-17:
Currently a user can specify

    --geopm-report-signals=TIME@cpu

to specify that they want the break down of time spent in each region by each CPU.  All signals of this sort are averaged (mean value), but it would be useful if a user could specify different aggregation functions like max, min, median or standard deviation.  Support for this feature may be provided by yet a more complicated GEOPM_REPORT_SIGNALS:

    --geopm-report-signals=TIME@cpu%max,TIME@cpu%stddev,TIME@cpu%median
    
would enable the user to collect the time with different Accumulator implementations.  Not sure '%' is the most appropriate delimiter for this.

dannosliwcd, 2021-06-17:
related: #1599 

I think the proposed solution in this issue would address the concern noted in that other issue. Particularly, once we have a way to key by agg function, it resolves the issue of having no way to distinguish multiple aggregators in SampleAggregator.

Some remaining questions to answer:

Right now, the mean aggregator automatically assumes weighted mean. Do we want to default to weighted stddev & weighed median, or regular stddev? I guess it depends on whether we want to answer questions like "we observed X over half the time" vs "we observed X over half the samples". 

How much are we willing to approximate a streaming median? Or do we assume an exact median is required?

fkeceli, 2021-06-17:
I think the weighted versions make more sense.