geopm/geopm #1767: Controller timeout with "Error: map::at"

fkeceli, 2021-07-23:
It is a mysterious error I have observed with certain combinations of geopmbench. Last one that gave me the error was:

> echo {\"loop-count\": 10, \"region\": [\"dgemm\", \"stream\"], \"big-o\": [8, 3]} > geopmbench_config.json
> geopmlaunch srun -N 1 -n 1 $(which geopmbench) geopmbench_config.json

Error: map::at will be observed, execution will continue and controller will timeout at the end.

fkeceli, 2021-07-23:
Steps to debug:

> geopmlaunch srun -N 1 -n 1 $(which geopmbench) geopmbench_config.json --geopm-debug-attach 0

Attach to the PID in another terminal. Start GDB and in GDB:

> attach <PID>
> break Controller.cpp:354
> set cont=1
> cont
> catch throw
> cont
> bt
> print $_exception

fkeceli, 2021-07-23:
To answer the comments about if this happens because of a out-of-mem problemlems with stream. It is a probl;em with small big-o:

```
echo {\"loop-count\": 10, \"region\": [\"dgemm\", \"stream\"], \"big-o\": [0.1, 0.1]} > geopmbench_config.json && geopmlaunch srun -N 1 -n 1 $(which geopmbench) geopmbench_config.json
```

fkeceli, 2021-07-24:
Checks at ApplicationStatus::get_hint and ApplicationStatus::set_hint are insufficient -- they don't check if the hint is one-hot. After this we still need to figure out where the invalid hint is coming from.

fkeceli, 2021-07-24:
Following two lines are from the trace file:
```
0.904473135|0|0x644f9787|0x100000000|0|441417.9901123047|70081.98922959137|185.0445125394556|10.70031922582649|2652272727.272727|42261895237976|75005617525044|44.97727272727273
0.909474859|0|0x644f9787|0x3800000000|0|441418.8278808594|70082.03908005524|185.6838435266437|10.73999058533855|2618181818.181818|42262475086524|75006053620032|44.90909090909091
```
0x644f9787 is model-init. Could be a due to a change in openmp or related to ompt.

fkeceli, 2021-07-24:
--disable-ompt does not change the error.

fkeceli, 2021-07-24:
A guess on what might be happening: In a recent PR we added some warmup runs to the DGEMM model app where some epochs run during the initialization phase. This might have exposed a bug in how hints are accumulated across nested regions.

bgeltz, 2021-07-26:
I'm seeing this in some of the integration tests as well:
http://mcfly.ra.intel.com/~test/cron_runs/2021-07-26_0000/test_power_balancer_power_governor_geopmbench.log
http://mcfly.ra.intel.com/~test/cron_runs/2021-07-24_0000/test_power_balancer_power_governor_geopmbench.log
...
http://mcfly.ra.intel.com/~test/cron_runs/2021-07-22_0000/test_power_balancer_power_governor_geopmbench.log

This has been failing since 7/22.  The run on 7/21 was fine:
http://mcfly.ra.intel.com/~test/cron_runs/2021-07-21_0000/test_power_balancer_power_governor_geopmbench.log

This PR which we all suspect is exposing the issue:
https://github.com/geopm/geopm/pull/1757
was merged on 7/21 at 11am.

Lastly, if I comment out the warmup loop that was introduced in 1757, the issue goes away.  

The integration test that is failing only tries to execute DGEMM, so debug can be simplified a bit since it doesn't appear to be related to shared memory corruption or aggregation issues.  I'm still digging.

bgeltz, 2021-07-26:
Minimal reproducer:

app.config
```
{
 "loop-count": 5,
 "region": ["dgemm-imbalance"],
 "big-o": [8.0],
 "imbalance": [0.5],
 "hostname": ["mcfly1"]
}
```

Monitor run:
```
$ OMP_NUM_THREADS=1 geopmlaunch srun -N1 -n1 -- geopmbench --verbose debug.config
```

Curiously if I remove the "-imbalance" and associated options from the app config, this all goes away.  I'm going to pick at that for a bit.

bgeltz, 2021-07-26:
```
(gdb) bt
#0  0x00007f38458e0420 in raise () from /lib64/libc.so.6
#1  0x00007f38458e1a01 in abort () from /lib64/libc.so.6
#2  0x00007f3845f1cfc6 in ?? () from /usr/lib64/libstdc++.so.6
#3  0x00007f3845f2877c in ?? () from /usr/lib64/libstdc++.so.6
#4  0x00007f3845f287e7 in std::terminate() () from /usr/lib64/libstdc++.so.6
#5  0x00007f3845f28a99 in __cxa_throw () from /usr/lib64/libstdc++.so.6
#6  0x00007f3845f1fa42 in std::__throw_out_of_range(char const*) () from /usr/lib64/libstdc++.so.6
#7  0x00007f385687ec62 in std::map<unsigned long, double, std::less<unsigned long>, std::allocator<std::pair<unsigned long const, double> > >::at (this=0x1f01300, __k=@0x1f0ee48: 249108103168)
    at /usr/include/c++/7/bits/stl_map.h:533
#8  0x00007f385686dbf7 in geopm::ApplicationSamplerImp::update (this=0x7f3856e0cd60 <instance>,
    curr_time=...) at src/ApplicationSampler.cpp:199
#9  0x00007f38568a43e4 in geopm::Controller::walk_up (this=0x1eea710) at src/Controller.cpp:459
#10 0x00007f38568a3b8e in geopm::Controller::step (this=0x1eea710) at src/Controller.cpp:403
#11 0x00007f38568a3960 in geopm::Controller::run (this=0x1eea710) at src/Controller.cpp:369
#12 0x00007f38568a0b4f in geopm_run_imp (ctl=0x1eea710) at src/Controller.cpp:119
#13 0x00007f38568a0c07 in geopm_ctl_run (ctl=0x1eea710) at src/Controller.cpp:131
#14 0x00007f3856a96f05 in _INTERNAL7fb94de2::geopm_pmpi_init (
    exec_name=0x7ffcbb2b5cbe "/home/test/build/geopm/bin/geopmbench")
    at src/geopm_pmpi_helper.cpp:171
#15 0x00007f3856a97425 in geopm_pmpi_init_thread (argc=0x7ffcbb2b3fc8, argv=0x7ffcbb2b3fd0,
    required=0, provided=0x7ffcbb2b3df0) at src/geopm_pmpi_helper.cpp:297
#16 0x00007f3856a925e8 in MPI_Init (argc=0x7ffcbb2b3fc8, argv=0x7ffcbb2b3fd0)
    at src/geopm_pmpi.c:65
#17 0x0000000000426140 in main (argc=3, argv=0x7ffcbb2b4328) at src/geopmbench_main.cpp:119
```

bgeltz, 2021-07-26:
I just put this together to try and isolate what is causing this OOR error:
https://github.com/bgeltz/geopm/commit/f8809f5eeb0317b109f4cd5d6db3db472e8a3ea5

My current debug procedure:
```
$ OMP_NUM_THREADS=1 geopmlaunch srun -N1 -n1 --geopm-debug-attach=0 -- geopmbench --verbose debug.config
```
Then, in another terminal window on the same host:
```
$ gdb-oneapi -p $(pgrep -al bench | head -n1 | cut -d' ' -f1)
...
(gdb) break src/ApplicationSampler.cpp:205
(gdb) set cont=1
(gdb) c
```
Which should break into the first rank (the Controller) when this OOR exception is first raised.  The second rank (the app) will continue onward, then hang/crash at the end of execution.  This is all expected since I've got the Controller paused in the debugger.

With the Controller stopped, I can look around a bit and see that the ```cpu_idx``` where the problem first  occurs is the last physical CPU before moving on to hyperthreads.  When the exception first occurs, the hint value that's being looked up is:
```
(gdb) p m_hint_last[cpu_idx]
$5 = (std::vector<uint64_t, std::allocator<uint64_t> >::value_type &) @0x2374a48: 249108103168
(gdb)
```

That hint value:
```
>>> hex(249108103168)
'0x3a00000000'
>>> bin(249108103168)
'0b11101000000000000000000000000000000000'
```
doesn't map to anything I expected it to.  Bits 33, 35, 36, and 37 are set.  I don't think multiple bits should ever be set any longer.  Looking at the hints for ```[cpu_idx - 1]```, only 1 bit is set.

I'm back to the theory that memory is being corrupted somewhere that's leading to this.

fkeceli, 2021-07-27:
That was our observation with Chris as well. I thought I saw such a behavior when we had multiple levels of nested regions. So I guessed somehow the hints are getting accumulated in a weird way.

bgeltz, 2021-07-30:
```
#0  geopm::ApplicationStatusImp::set_hint (this=0x1228510, cpu_idx=43,
    hint=249108103168) at src/ApplicationStatus.cpp:99
#1  0x00007f76b2175726 in geopm::ProfileImp::set_hint (this=0x7f76b24f5ea0 <instance>,
    hint=249108103168) at src/Profile.cpp:631
#2  0x00007f76b2172b9c in geopm::ProfileImp::enter (this=0x7f76b24f5ea0 <instance>,
    region_id=4060874923062537575) at src/Profile.cpp:456
#3  0x00007f76b21679bd in geopm_prof_enter (region_id=4060874923062537575)
    at src/DefaultProfile.cpp:120
#4  0x0000000000414aa1 in geopm::ModelRegion::region_enter (this=0x125e6c0)
    at src/ModelRegion.cpp:176
#5  0x000000000040fcbd in geopm::DGEMMModelRegion::run (this=0x125e6c0)
    at src/DGEMMModelRegion.cpp:148
#6  0x000000000040f760 in geopm::DGEMMModelRegion::big_o (this=0x125e6c0, big_o_in=8)
    at src/DGEMMModelRegion.cpp:136
#7  0x000000000040ef9f in geopm::DGEMMModelRegion::DGEMMModelRegion (this=0x125e6c0,
    big_o_in=8, verbosity=0, do_imbalance=true, do_progress=false, do_unmarked=false)
    at src/DGEMMModelRegion.cpp:87
#8  0x000000000041a2d9 in geopm::make_unique<geopm::DGEMMModelRegion, double&, int&, bool&, bool&, bool&> (args=@0x7ffc9ad771f2: false, args=@0x7ffc9ad771f2: false,
    args=@0x7ffc9ad771f2: false, args=@0x7ffc9ad771f2: false,
    args=@0x7ffc9ad771f2: false) at src/Helper.hpp:49
#9  0x0000000000413c28 in geopm::ModelRegion::model_region (name=..., big_o=8,
    verbosity=0) at src/ModelRegion.cpp:86
#10 0x00000000004108e5 in geopm::ModelApplication::ModelApplication (
    this=0x7ffc9ad778d8, repeat=5, region_name=..., big_o=..., verbosity=0, rank=0)
    at src/ModelApplication.cpp:59
#11 0x0000000000426be4 in main (argc=2, argv=0x7ffc9ad77b08)
    at src/geopmbench_main.cpp:176
```

bgeltz, 2021-07-30:
This is an uninitialized memory problem.  In the ```DGEMMModelRegion``` constructor, we're calling into ```big_o``` on line 87 before ```ModelRegion::region()`` is ever called which happens on line 88.  Because the warmup loop is in there now which calls ```run()``` with an uninitialized region_id, we start to get into a bad situation.

I'm going to add some exceptions tomorrow to ensure ```region()``` has completed before ```run()``` is ever called. 

I think the simple solution to this bug is just to move the ```big_o``` call to the end of the constructor.  More testing tomorrow.

bgeltz, 2021-08-04:
The patch to fix this and other associated clean-up is in this PR:
https://github.com/geopm/geopm/pull/1769

commit b45bf058 (2021-08-06): Move timing of DGEMM warmup loop

- Ensure prof_region() has been called before
  prof_enter().
- Fixes #1767.
closed 2021-08-06 by bgeltz