geopm/geopm #1874: Spack and geopm

maiterth, 2021-09-17:
I got a ping on geopm usage with spack on a cray machine.

Installation issue: geopm #26003 
https://github.com/spack/spack/issues/26003

Spack integration with geopm was a task on one of the projects with LLNL, at the time I took on the task. 
After having the initial version of geopm ( up to geopm1.1) on there working, I have not looked into this.

Since this is not the primary (or even recommended mode of installation according to geopm's readme) I'm not sure if this is a priority. But it may be an interesting platform, therefore figuring out the issue might be interesting to engage with.

Thanks,
--Matthias

cmcantalupo, 2021-09-17:
Thanks for bringing this to our attention @maiterth!  I have left a  not on the spack issue.

cmcantalupo, 2021-09-17:
#1875

cmcantalupo, 2023-12-13:
Done criteria:
- [ ] Working recipes for all packages: covering all installed content from geopm service and base directory
- [ ] Recipes support building from release tag and also from an un-tagged version of the dev branch
- [ ] Pull request has been submitted to upstream Spack repo

closed 2024-10-30 by bgeltz