geopm/geopm #2028: test_ee_short_region_slop / test_fmap_shor_region_slop fails to parse trace files

cmcantalupo, 2021-12-03:
**Describe the bug**
I tried to run the test_ee_short_region_slop.py test and I expected plots to be generated instead an error was generated when the trace file was parsed.

**GEOPM version**
73a7ec4ae2bb9e68cb91a2d1c8151346e6148f1f

**Expected behavior**
The test_ee_short_region_slop.py is misnamed, it does not use the energy efficient agent, but rather the frequency map agent.  This is not a test, so much as an analysis tool that generates visualizations, and as such it is not run in our nightly tests.  

**Actual behavior**
When this script was executed the parser found "NAN" values in the first column of the trace.  This breaks the parser, and the plots are not generated.  

**Error messages produced**
```
======================================================================
ERROR: test_generate_report_plot (__main__.TestIntegration_ee_short_region_slop)
Test to visualize the data in the report files
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/nomsr/.local/lib/python3.6/site-packages/pandas/core/indexes/base.py", line 2898, in get_loc
    return self._engine.get_loc(casted_key)
  File "pandas/_libs/index.pyx", line 70, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 101, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 1675, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 1683, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'duration (sec)'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./test_ee_short_region_slop.py", line 228, in test_generate_report_plot
    generate_report_plot(report_df, out_path)
  File "./test_ee_short_region_slop.py", line 361, in generate_report_plot
    generate_report_subplot(report_df, rn, 'duration (sec)', ya)
  File "./test_ee_short_region_slop.py", line 379, in generate_report_subplot
    xdata = selected_data.mean()[xaxis]
  File "/home/nomsr/.local/lib/python3.6/site-packages/pandas/core/frame.py", line 2906, in __getitem__
    indexer = self.columns.get_loc(key)
  File "/home/nomsr/.local/lib/python3.6/site-packages/pandas/core/indexes/base.py", line 2900, in get_loc
    raise KeyError(key) from err
KeyError: 'duration (sec)'
```

```
======================================================================
ERROR: test_generate_trace_plot (__main__.TestIntegration_ee_short_region_slop)
Test to visualize the time series data in the trace files
----------------------------------------------------------------------
Traceback (most recent call last):
  File "./test_ee_short_region_slop.py", line 249, in test_generate_trace_plot
    trace = read_trace(in_path)
  File "./test_ee_short_region_slop.py", line 404, in read_trace
    return pandas.read_csv(path, sep='|', skiprows=skiprows, converters=converters)
  File "/home/nomsr/.local/lib/python3.6/site-packages/pandas/io/parsers.py", line 688, in read_csv
    return _read(filepath_or_buffer, kwds)
  File "/home/nomsr/.local/lib/python3.6/site-packages/pandas/io/parsers.py", line 460, in _read
    data = parser.read(nrows)
  File "/home/nomsr/.local/lib/python3.6/site-packages/pandas/io/parsers.py", line 1198, in read
    ret = self._engine.read(nrows)
  File "/home/nomsr/.local/lib/python3.6/site-packages/pandas/io/parsers.py", line 2157, in read
    data = self._reader.read(nrows)
  File "pandas/_libs/parsers.pyx", line 847, in pandas._libs.parsers.TextReader.read
  File "pandas/_libs/parsers.pyx", line 862, in pandas._libs.parsers.TextReader._read_low_memory
  File "pandas/_libs/parsers.pyx", line 941, in pandas._libs.parsers.TextReader._read_rows
  File "pandas/_libs/parsers.pyx", line 1051, in pandas._libs.parsers.TextReader._convert_column_data
  File "pandas/_libs/parsers.pyx", line 2139, in pandas._libs.parsers._apply_converter
  File "./test_ee_short_region_slop.py", line 396, in <lambda>
    converters={'REGION_HASH': lambda xx: int(xx, 16)}
ValueError: invalid literal for int() with base 16: 'NAN'
```

**Additional context**
This was found when testing the service, in this case all of the test_*.py files were executed rather than relying on the imports in `geopm/integration/test/__main__.py`.

bgeltz, 2021-12-14:
This test is not run as a part of the nightlies, so the first issue ```duration (sec)``` is related to when we moved to use proper SI units everywhere.  This was not updated.

The second issue is a little more nuanced, but is related to changes that came in with this PR: 
https://github.com/geopm/geopm/pull/1496
Specifically commit a6d65e06fb
That commit was meant to address https://github.com/geopm/geopm/issues/1070 and https://github.com/geopm/geopm/issues/1368.
The side effect of that commit is that now the very first entry in all trace files is a NAN for the region hash and hint.  This test breaks when that behavior occurs.

bgeltz, 2021-12-16:
I've been working through these issues on my issue-2015 branch:
https://github.com/bgeltz/geopm/commits/issue-2015

The following patches address the above issues:
https://github.com/bgeltz/geopm/commit/d35c168d3ecb0fe9b9da44d0052f497b31ad62f5
https://github.com/bgeltz/geopm/commit/672cf470af4670726080bbbb6b1a7283ee4abe0c

However they expose a new issue.  In the PR I mentioned in my previous comment (#1496), the ProfileIOGroup was modified to remove the `REGION_COUNT` signal.  This test was never updated to account for that change.

Additionally, it's unclear how we will be able to derive `REGION_COUNT` signal presently.  It appears as though `REGION_PROGRESS` is not indicating region exit.  In a trace I have in pandas, pandas reports that the only values are:
```
>>> trace['REGION_PROGRESS'].unique()
array([nan,  0.])
```

bgeltz, 2021-12-17:
Since this is broken in dev and is not specifically related to test failing when msr-safe permissions are removed, this is tech-debt.  This should be revisited when the tests that used to rely on the ```--enable-bloat``` features (e.g. injecting a 1 in the `REGION_PROGRESS` field at every region exit) are rewritten to use the profile trace instead.

All of the old signals re:bloat can be derived from the profile trace.

bgeltz, 2021-12-17:
This was accidentally closed.

commit 24591111 (2021-12-17): Use proper units for slop test

- This test is not run with the nightlies.
- Resolves #2028.