geopm/geopm #2117: Group directory used by AccessLists does not seem to be securely created in all cases

cmcantalupo, 2022-02-16:
**Describe the bug**
After installing a new version of the service, the 0.DEFAULT_ACCESS directory has group and world read and execute permissions

**GEOPM version**
Output from `geopmread --version`
690d51a

**Expected behavior**
Directory should be 0o700 mode.

**Actual behavior**
Mode is 0o755

**Error messages produced**
None

**Additional context**

cmcantalupo, 2022-03-16:
Install of new version of geopm that created directory with reduced privilege was run after the directory was previously created insecurely.  We saw that the directory retained its excessive privilege.

bgeltz, 2022-03-23:
I was able to confirm that if ```/etc/geopm-service/0.DEFAULT_ACCESS``` was left over from a previous install, a subsequent install of the service then issuing `geopmaccess -g ''` will successfully return all the signals/controls in the directory.  No error or warning is produced.

Calling into `AccessLists.get_group_access()` in `system_files.py` never validates the permissions on the directory in question. `secure_read_file()` is ultimately called from the `AccessLists._read_allowed()` helper, and that *will* verify the permissions on the `allowed_signals` and `allowed_controls` files.  But nothing ever verifies the permissions on the directories those files are sitting in.

I think we need to add `is_secure_dir` similar to `is_secure_path` for this case.

PR #2209 (2022-03-28): Ensure access lists dirs are created securely before use
Resolves #2117.
commit fc1d2ac6 (2022-03-28): Ensure service directories are secure before use

- Fixes #2117.
closed 2022-03-28 by cmcantalupo