geopm/geopm #2135: As a user of the GEOPM reports I would like a system totals section

lhlawson, 2022-03-05:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I add GPU signals to the report and it reports a value of 0 due to the aggregation being based on which CPU the application was running on (i.e. if the CPU pinning does not match the 'ideal' affinitization from PlatformTopo/AcceleratorTopo).  
Example from a PARRES DGEMM run:
`    Application Totals:
      runtime (s): 15.0059
      count: 0
      sync-runtime (s): 15.0059
      package-energy (J): 3724.23
      dram-energy (J): 683.414
      power (W): 248.183
      frequency (%): 129.166
      frequency (Hz): 3.09997e+09
      time-hint-network (s): 0.0258852
      time-hint-ignore (s): 0
      time-hint-compute (s): 0
      time-hint-memory (s): 0
      time-hint-io (s): 0
      time-hint-serial (s): 0
      time-hint-parallel (s): 0
      time-hint-unknown (s): 0
      time-hint-unset (s): 14.9791
      gpu-energy (J): 13009.4
      gpu-power (W): 868.16
      gpu-frequency (Hz): 1.53e+09
      GPU_ENERGY@board_accelerator-0: 0
      GPU_ENERGY@board_accelerator-1: 0
      GPU_ENERGY@board_accelerator-2: 0
      GPU_ENERGY@board_accelerator-3: 3160.22
      ENERGY_PACKAGE@package-0: 0
      ENERGY_PACKAGE@package-1: 2051.83
      geopmctl memory HWM (B): 228688
      geopmctl network BW (B/s): 0
`

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

The addition of a System totals section to the report that provides data on the system level behavior regardless of application.  Example below: 
`
    System Totals:
      runtime (s): 15.0059
      package-energy (J): 3724.23
      dram-energy (J): 683.414
      power (W): 248.183
      frequency (%): 129.166
      frequency (Hz): 3.09997e+09
      gpu-energy (J): 13009.4
      gpu-power (W): 868.16
      gpu-frequency (Hz): 1.53e+09
      GPU_ENERGY@board_accelerator-0: 3860.22
      GPU_ENERGY@board_accelerator-1: 3760.22
      GPU_ENERGY@board_accelerator-2: 3660.22
      GPU_ENERGY@board_accelerator-3: 3460.22
      ENERGY_PACKAGE@package-0: 1673.8
      ENERGY_PACKAGE@package-1: 2051.83
`

**Describe alternatives you've considered**
Decoupling GPU reporting from CPUs entirely (significant dev effort), simply using the GEOPM traces and ignoring the reports (tracing may impact workload perf), using the existing 'gpu-energy (J)' reading only (isn't as useful for single GPU workloads).  

**Additional context**
Application total provides a count value of 0, which should probably be removed.

cmcantalupo, 2022-12-20:
```
geopmlaunch srun -N1 -n1 --geopm-affinity-disable --cpu-bind=map_cpu=50 --geopm-report-signals=CPU_ENERGY@package ../Git/geopm/examples/print_affinity 
```

```
GEOPM Version: 2.0.0+dev327g4fdf753b3
Start Time: Tue Dec 20 10:36:58 2022
Profile: "/home/cmcantal/Git/geopm/examples/.libs/print_affinity"
Agent: monitor
Policy: {}

Hosts:
  mcfly1:
    Regions:
    -
      region: "[OMPT]main+0x534"
      hash: 0x7c084d36
      runtime (s): 0.263207
      count: 1
      sync-runtime (s): 0.26052
      package-energy (J): 32.3127
      dram-energy (J): 2.76048
      power (W): 124.032
      frequency (%): 151.695
      frequency (Hz): 3.1856e+09
      time-hint-network (s): 0
      time-hint-ignore (s): 0
      time-hint-compute (s): 0
      time-hint-memory (s): 0
      time-hint-io (s): 0
      time-hint-serial (s): 0
      time-hint-parallel (s): 0
      time-hint-unknown (s): 0.260002
      time-hint-unset (s): 0
      time-hint-spin (s): 0
      uncore-frequency (Hz): 2.4e+09
      CPU_ENERGY@package-0: 16.6171
      CPU_ENERGY@package-1: 0
    -
      region: "MPI_Barrier"
      hash: 0x7b561f45
      runtime (s): 4.388e-06
      count: 1
      sync-runtime (s): 0
      package-energy (J): 0
      dram-energy (J): 0
      power (W): 0
      frequency (%): 0
      frequency (Hz): 0
      time-hint-network (s): 0
      time-hint-ignore (s): 0
      time-hint-compute (s): 0
      time-hint-memory (s): 0
      time-hint-io (s): 0
      time-hint-serial (s): 0
      time-hint-parallel (s): 0
      time-hint-unknown (s): 0
      time-hint-unset (s): 0
      time-hint-spin (s): 0
      CPU_ENERGY@package-0: 0
      CPU_ENERGY@package-1: 0
    Unmarked Totals:
      runtime (s): 0.00234991
      count: 0
      sync-runtime (s): 0
      package-energy (J): 0
      dram-energy (J): 0
      power (W): 0
      frequency (%): 0
      frequency (Hz): 0
      time-hint-network (s): 0
      time-hint-ignore (s): 0
      time-hint-compute (s): 0
      time-hint-memory (s): 0
      time-hint-io (s): 0
      time-hint-serial (s): 0
      time-hint-parallel (s): 0
      time-hint-unknown (s): 0
      time-hint-unset (s): 0
      time-hint-spin (s): 0
      uncore-frequency (Hz): 0
      CPU_ENERGY@package-0: 0
      CPU_ENERGY@package-1: 0
    Epoch Totals:
      runtime (s): 0
      count: 0
      sync-runtime (s): 0
      package-energy (J): 0
      dram-energy (J): 0
      power (W): 0
      frequency (%): 0
      frequency (Hz): 0
      time-hint-network (s): 0
      time-hint-ignore (s): 0
      time-hint-compute (s): 0
      time-hint-memory (s): 0
      time-hint-io (s): 0
      time-hint-serial (s): 0
      time-hint-parallel (s): 0
      time-hint-unknown (s): 0
      time-hint-unset (s): 0
      time-hint-spin (s): 0
      uncore-frequency (Hz): 0
      CPU_ENERGY@package-0: 0
      CPU_ENERGY@package-1: 0
    Application Totals:
      runtime (s): 0.265561
      count: 0
      sync-runtime (s): 0.265561
      package-energy (J): 32.9574
      dram-energy (J): 2.81554
      power (W): 124.105
      frequency (%): 151.919
      frequency (Hz): 3.1903e+09
      time-hint-network (s): 0
      time-hint-ignore (s): 0
      time-hint-compute (s): 0
      time-hint-memory (s): 0
      time-hint-io (s): 0
      time-hint-serial (s): 0
      time-hint-parallel (s): 0
      time-hint-unknown (s): 0.260002
      time-hint-unset (s): 0
      time-hint-spin (s): 0
      uncore-frequency (Hz): 2.4e+09
      CPU_ENERGY@package-0: 16.9453
      CPU_ENERGY@package-1: 0
      geopmctl memory HWM (B): 89296
      geopmctl network BW (B/s): 0
```

cmcantalupo, 2022-12-20:
Simple reproducer posted here:

https://github.com/cmcantalupo/geopm/tree/app-totals-test

closed 2023-07-11 by cmcantalupo