geopm/geopm #2161: The "make rpm" target fails to build on the OpenSUSE 15.3 OBS distribution

dannosliwcd, 2022-03-16:
I tried packaging geopm for OpenSUSE 15.3 under OpenSUSE Build Service, and encountered a few errors;

* python3 fails to import setuptools (need an explicit python3-setuptools build dependency)
* dbus-related directories are not associated with any packages (missing libdbus-1-3 dependency)
* Even after adding libdbus, the `interfaces` directory (in `/etc`) is still orphaned. However, it looks like there is one from libdbus in `/usr/share`. From the dbus documentation, it seems like package maintainers are expected to place their own configurations in /usr/share, and admins should place their overrides in /etc.

This is on the current head of dev after the fix for #2159 (at 13f198aceea32dc1729b5762732ab3be7be5c7cf).

dannosliwcd, 2022-03-16:
I came across this as well: https://fedoraproject.org/wiki/Changes/Reduce_dependencies_on_python3-setuptools

So the explicit setuptools dependency might be useful for additional variants, rather than just suse-like ones (possibly rhel-like variants if this Fedora 35 change makes it in to them).

dannosliwcd, 2022-03-16:
Also, it looks like RH-like packaging guidelines recommend removing group tags. SUSE-like guidelines recommend using them.

There is a draft example of the suggested changes: dannosliwcd/geopm#1

cmcantalupo, 2022-03-18:
Related to #2178

dannosliwcd, 2022-03-22:
The changes in #2179 include fixes for this issue. #2188 adds OBS builds to our CI workflow.

closed 2022-03-22 by dannosliwcd