geopm/geopm #2168: Python dependency requirements are not consistent and may be out of date

dannosliwcd, 2022-03-16:
The requirements.txt and setup.py dependencies in geopmdpy do not match. This isn't necessarily a problem (e.g., if requirements is meant for a different build/test/run use case from setup.py), but the setup version does look insufficient.

The required dasbus versions do not match. I am aware of a 1.5.0+ requirement since they added a feature to help us with #1606. I'm not sure if our later changes add a 1.6 requirement.

Sphinx is missing from the setup.py requirements, which can result in documentation build failures.

Current geopmdpy requirements.txt:
```
cffi>=1.13.2
dasbus>=1.6
PyGObject>=3.34.0
psutil>=5.4.8
docstring_parser>=0.8.1
sphinx>=4.1.0
sphinx_rtd_theme>=0.5.2
jsonschema>=2.6.0
```

setup.py:
```
                    'cffi>=1.6.0',
                    'setuptools>=39.2.0',
                    'future>=0.17.1',
                    'psutil>=5.6.2',
                    'dasbus>=1.5.0',
                    'jsonschema>=2.6.0'
```

geopmpy requirements.txt:
```
matplotlib>=2.2.2
cycler>=0.10.0
numpy>=1.14.3
setuptools>=39.2.0
natsort>=5.3.2
psutil>=5.4.8
pandas>=0.23.0
tables>=3.4.3
cffi>=1.6.0
pyyaml>=5.1.0
future>=0.17.1
```

setup.py
```
                    'pandas>=0.23.0',
                    'natsort>=5.3.2',
                    'matplotlib>=2.2.2',
                    'cycler>=0.10.0',
                    'tables>=3.4.3',
                    'psutil>=5.4.8',
                    'cffi>=1.6.0',
                    'numpy>=1.14.3',
                    'setuptools>=39.2.0',
                    'pyyaml>=5.1.0',
                    'future>=0.17.1',
                    'geopmdpy'
```

Pandas upgraded to 1.0 a while ago. We were previously blocked from upgrading that dependency since we were concurrently supporting python2 and python3 (and 1.0 is when they droppped python2 support). We are able to upgrade now that we dropped python 2 support. We should probably modify our dependency to be a stable release series since they exist now.

dannosliwcd, 2022-03-17:
See https://geopm.github.io/requires.html for our documented requirements.

dannosliwcd, 2022-03-28:
After #2208, the 'future>=0.17.1' requirement can be removed, too.

dannosliwcd, 2022-04-06:
It could also be worth documenting that our gobject dependency has a nested dependency on pycairo by default, and that it may fail to build for a user under some of our target platforms. Generally that's avoidable by just installing a python3-gobject package from the system package manager, but that won't be an option for non-root users. This can be worked around by running `PYGOBJECT_WITHOUT_PYCAIRO=1 python3 -m pip install pygobject` as noted in https://gitlab.gnome.org/GNOME/pygobject/-/issues/395

bgeltz, 2022-04-25:
I couldn't get Daniel's suggestion on installing `pygobject` without cairo via pip to work.  It looks like pip changed their handling around this: https://github.com/pygobject/pycairo/issues/219#issuecomment-732320682

I had to use:
```
PYGOBJECT_WITHOUT_PYCAIRO=1 python3 -m pip install --no-build-isolation --no-use-pep517 pygobject
```
But I was still not successful:
```
... lots of build output
      copying gi/overrides/__init__.py -> build/lib.linux-x86_64-cpython-38/gi/overrides
      running build_ext
      Package gobject-introspection-1.0 was not found in the pkg-config search path.
      Perhaps you should add the directory containing `gobject-introspection-1.0.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'gobject-introspection-1.0' found
      Command '('pkg-config', '--print-errors', '--exists', 'gobject-introspection-1.0 >= 1.56.0')' returned non-zero exit status 1.

      Try installing it with: 'sudo apt install libgirepository1.0-dev'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> pygobject
```

In the end I think it's best to just get our build time dependencies installed rather than work around them.  It's problematic and error prone.

bgeltz, 2022-05-11:
Issues remain in the scripts directory around this.

PR #2306 (2022-05-12): Update geopmpy requirements to current for v3.6.15
- Fixes #2168.
- Related to #2257.
- Related to #2284.
commit b24efbf4 (2022-05-12): Update geopmpy requirements to current for v3.6.15

- Alphabetize both lists.
- Fixes #2168.

Change-Id: I463f9e6bff4bfb6e86d8a4653a25caf5fe470c9c
closed 2022-05-12 by cmcantalupo