geopm/geopm #2169: Power control MSR aliases are missing signal aliases to read the current control values

dannosliwcd, 2022-03-16:
**Describe the bug**
The `POWER_PACKAGE_LIMIT` control is an alias for `MSR::PKG_POWER_LIMIT:PL1_POWER_LIMIT`. There is no equivalent alias for a signal. Such an alias would be useful to check the current power limit.

Expected behavior: `POWER_PACKAGE_LIMIT` should also be a signal alias for `MSR::PKG_POWER_LIMIT:PL1_POWER_LIMIT`

Similarly, `POWER_PACKAGE_TIME_WINDOW` should be a signal alias for `MSR::PKG_POWER_LIMIT:PL1_TIME_WINDOW`

**GEOPM version**
1.1.0+dev1800g03ef56b2

**Alternatives to consider**
This may be a good opportunity to add aliases following the new format: `<HIGH_LEVEL_DOMAIN>_<NAME>_<SPECIALIZATION>_[CONTROL]`

bgeltz, 2022-03-16:
https://github.com/geopm/geopm/blob/dev/service/src/MSRIOGroup.cpp#L221
```
$ geopmread POWER_PACKAGE_LIMIT board 0
140
$ geopmread POWER_PACKAGE_TIME_WINDOW board 0
1
```

Are you using an old version of GEOPM?  These should be in place already.

closed 2022-03-16 by bgeltz

dannosliwcd, 2022-03-16:
Confirmed. Sorry for the false alarm.