geopm/geopm #2212: As a GEOPM service administrator I would like better support for validating access

cmcantalupo, 2022-03-28:
**Is your feature request related to a problem? Please describe.**
It is common to not be able to write to /etc within the same environment where the service is running.  An example would be having /etc be mounted to support multiple runtime environments.  

I'm always frustrated when I cannot modify /etc/geopm-service within the runtime environment, but have no way to validate files in /etc/geopm-service have been configured correctly, or validate that a request to change them is valid without actually modifying /etc. 

**Describe the solution you'd like**
Provide command line options to geopmaccess to support:
    a. Verify that all of the files in /etc/geopm-service are named correctly, and that all of the signals and controls listed in these files are valid.
    b. Add a "dry-run" option to geopmaccess that will not modify files in /etc/geopm-service, but will tell the administrator if the request to modify an access list is valid.

**Describe alternatives you've considered**
As it is now, an administrator must validate access lists created in a different environment by hand.

**Additional context**
Thanks @bensallen for the suggestion.

bensallen, 2022-03-29:
Each of these verification types (a) and (b) should support two modes:

1) validate against the running system's discovered valid controls and signals.
2) validate against geopm's list of all know valid controls and signals, but do not consider those discovered on the running system. This more lax validation mode will be useful during an OS image build process that likely will occur on unrelated hardware and perhaps even distro or kernel version as the compute nodes.

cmcantalupo, 2022-04-19:
Make sure that all control names are valid signals in verification of access lists.

PR #2304 (2022-05-17): Update geopmaccess man page and CLI
- Fixes #2212
- Relates to #2198
- Could have better negative testing for incompatible command line option errors.
- Does not implement the -d option described in #2198 to remove a subset of names from an access list, added --edit mode instead
- Solution for #2212 is not exactly as described in the issue, but hopefully is sufficient.
closed 2022-05-17 by cmcantalupo