geopm/geopm #2338: Update MSRIOGroup aliases to provide a one to many alias

lhlawson, 2022-05-24:
> estimate 1
Parent [feature/story]: #XXXX

The implementation for the alias feature needs to be modified because it does not allow for a one to many mapping.  Both HWP and the Uncore frequency control aliases would benefit from being able to provide:
CPU_UNCORE_FREQUENCY_CONTROL --> MSR::UNCORE_RATIO_LIMIT:MIN_RATIO = MSR::UNCORE_RATIO_LIMIT:MAX_RATIO
CPU_CORE_FREQUENCY_CONTROL --> MSR::HWP_REQUEST:MINIMUM_PERFORMANCE = MSR::HWP_REQUEST:MAXIMUM_PERFORMANCE

Done criteria:

- [ ] Integration tests that cover the feature run and pass
- [ ] Dependencies mocked in unit tests are still valid
- [ ] Documentation changes are required since this change modifies interfaces (aliases)
- [ ] Documentation changes are required since this change adds interfaces (uncore aliases)

bgeltz, 2022-08-01:
Per this morning's discussion:

Change ```CPU_CORE_FREQUENCY_CONTROL``` to ```CPU_CORE_FREQUENCY_CONTROL_MAX``` and remove other aliases.  For P-states, we'll only have ```CONTROL_MAX``` exposed.  For HWP, we'll have ```CONTROL_MAX``` and ```CONTROL_MIN```.

bgeltz, 2022-08-12:
The PR that is related to this removes the ambiguity by establishing that all frequency controls are specified as a pair of MIN and MAX.

This should remain open until we implement the unified control or decide we do not want this feature.

commit 4130bd20 (2022-08-12): Rename CPU_FREQUENCY_CONTROL to CPU_FREQUENCY_MAX_CONTROL

- Resolves #2338.
- Purge GPU_CORE_FREQUENCY_CONTROL
- Add retry logic to NVMLIOGroup
- Updated scaling, min/max swizzle, save/restore
- Fixed min/max scaling in NAN case
- Replace control pruning code with EUID check for root
- Add is_privileged_access to NVMLDevicePool
- Fixed issues caused by NAN min/max freq control during PlatIO push control, trimmed dead code
- Min/Max control changes & retry loop for Level Zero
- Added write control for high level min/max L0 aliases
- Add all expected function calls to SetUp
- Statically allocat [... cut]