geopm/geopm #2382: POSIXSignalTest.sig_queue_EPERM test failing

Alessandro-Barbieri, 2022-06-16:
**Describe the bug**
I tried to run the tests of geopmd and I expected them to pass instead one failed

**GEOPM version**
2.0.0+rc1

**Expected behavior**
All tests passing

**Actual behavior**
1 test is failing

**Error messages produced**
FAIL: test/gtest_links/POSIXSignalTest.sig_queue_EPERM
Was debug messaging was enabled at build time? No

**Additional context**

see https://bugs.gentoo.org/852260
log https://852260.bugs.gentoo.org/attachment.cgi?id=785567

Alessandro-Barbieri, 2022-06-21:
Testing on my machine
```
cat /var/tmp/portage/sys-power/geopmd-2.0.0_rc1/work/geopm-2.0.0-rc1/service/test/gtest_links/POSIXSignalTest.sig_queue_EPERM.log
Note: Google Test filter = POSIXSignalTest.sig_queue_EPERM
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from POSIXSignalTest
[ RUN      ] POSIXSignalTest.sig_queue_EPERM
test/POSIXSignalTest.cpp:205: Failure
Expected to throw, but succeeded.
[  FAILED  ] POSIXSignalTest.sig_queue_EPERM (0 ms)
[----------] 1 test from POSIXSignalTest (0 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (0 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] POSIXSignalTest.sig_queue_EPERM
```

cmcantalupo, 2022-06-21:
Hi @Alessandro-Barbieri,

Thanks again for the feedback.  I believe that you are running this test as the "root" user.  The test that is failing is not valid when run by the "root" user.  The PR #2387 modifies the test to be skipped at runtime in this case.  Please let us know if this patch may be applied to the build to resolve your issue:

593caaa68a405e412506138a5ab2987c56d2ab06

The change will be included in the next release candidate.

cmcantalupo, 2022-06-29:
It seems that we will have to check for CAP_KILL in procfs rather than checking for user "root" with geteuid().  We have similar python code to check for CAP_SYSADMIN here:

https://github.com/geopm/geopm/blob/dev/service/geopmdpy/service.py#L838-L848

We will have to do something similar in C++ to check for CAP_KILL for the test process and skip the test if the capability is present.

Krebrov001, 2022-06-29:
The error is occuring because the `portage` package manager creates it's own user, which is not the `root` or `sudo` user, but contains elevated permissions.

Krebrov001, 2022-07-02:
@Alessandro-Barbieri  I have written a patch for this issue. Please download the revision of the code in the `dev` branch, the commit hash `2e958f3`. Please test, if the new code solves your issue, please let us know.

Krebrov001, 2022-07-12:
@Alessandro-Barbieri  Please check to see if the latest geopm release candidate fixes your issue.

PR #2387 (2022-06-21): Skip permission error test when running as root user
- The test POSIXSignalTest.sig_queue_EPERM sends a
  SIGCONT to a root owned process, and checks that
  an exception is raised.
- This is not a valid test when run as the root user
  so this patch skips the test and warns the user in
  that case.
- Fixes #2382
commit 593caaa6 (2022-06-21): Skip permission error test when running as root user

- The test POSIXSignalTest.sig_queue_EPERM sends a
  SIGCONT to a root owned process, and checks that
  an exception is raised.
- This is not a valid test when run as the root user
  so this patch skips the test and warns the user in
  that case.
- Fixes #2382
closed 2022-07-19 by cmcantalupo