geopm/geopm #2394: Address integer overflow issues identified by CodeQL

bgeltz, 2022-06-22:
> estimate 1
Parent story: #2219

Context:
PR introducing CodeQl scans: https://github.com/geopm/geopm/pull/2266
CodeQL report: https://github.com/geopm/geopm/pull/2266/checks?check_run_id=6118978583

Identifications:
1. service/src/LevelZeroAcceleratorTopo.cpp, line 65
2. src/EditDistPeriodicityDetector.cpp, line 173 - 174
3. googletest code is filtered out.

Done criteria:

- [ ] CodeQL report comes back clean.
- [ ] PR #2266 is merged.

Krebrov001, 2022-06-29:
@bgeltz By the way, what is the theoretical max number of the GPUs on these platforms?

```cpp
unsigned int num_gpu = m_levelzero_device_pool.
                               num_gpu(GEOPM_DOMAIN_GPU);
unsigned int num_gpu_chip = m_levelzero_device_pool.
                                    num_gpu(GEOPM_DOMAIN_GPU_CHIP);
```

bgeltz, 2022-06-29:
Certainly less than 2^32. A uint8 or uint16 should be future proof.