geopm/geopm #2463: GEOPM_SHMKEY environment variable feature should be removed

cmcantalupo, 2022-07-19:
This feature is confusing at best, and may cause failures in the attach process.  We have resolved issues around stale keys that this feature was originally a work around for.

Krebrov001, 2022-07-23:
@cmcantalupo Do we remove the `GEOPM_SHMKEY` environment variable only, or do we also remove the `--geopm-shmkey` and all features pertaining to shared memory keys?

Krebrov001, 2022-07-24:
`--geopm-shmkey key`

 Specify a prefix to be used with all the shared memory keys
 generated by the GEOPM runtime for communication with the
 application.  This prefix is not used for the endpoint.
 This is useful for avoiding collisions with keys
 that were not properly cleaned up.  The default key prefix
 is `"geopm-shm"`.  A shared memory key must have no
 occurrences of the `'/'` character.  The base key is used as
 the prefix for each shared memory region used by the
 runtime.  If the keys are left behind, a simple command to
 clean up after an aborted job is:

``` console
$ test -n "$GEOPM_SHMKEY" && rm -f /dev/shm${GEOPM_SHMKEY}* || rm -f /dev/shm/geopm-shm*
```
 This option is used by the launcher to set the
 `GEOPM_SHMKEY` environment variable.  The command line
 option will override any value currently set in the
 environment

Krebrov001, 2022-07-24:
> This option is used by the launcher to set the `GEOPM_SHMKEY` environment variable. The command line option will override any value currently set in the environment

If we want to remove the `GEOPM_SHMKEY` environment variable, then it stands that we also want to remove the `--geopm-shmkey` command line option?

PR #2488 (2022-07-29): Issue 2463
- Relates to #2463
- Fixes #2463

`GEOPM_SHMKEY` environment variable feature should be removed #2463
closed 2022-07-29 by cmcantalupo