geopm/geopm #2770: GEOPM runtime attempts to report GPU energy when no GPU topology information is available

dannosliwcd, 2022-12-20:
**Describe the bug**
geopm report files attempt to include gpu energy totals in a report when a GPU_ENERGY signal is available. However, PlatformIO also checks the signal topology as part of pushing a signal. If no topology information is available, then geopmlaunch fails to launch GEOPM.

**GEOPM version**
2.0.0+dev163g0b72a374a

**Expected behavior**
`geopmlaunch` generates a report with GPU energy if it is possible to get GPU energy information from GEOPM batch reads, and omits GPU energy otherwise.

**Actual behavior**
geopmlaunch returns an error because the reporter fails to push GPU_ENERGY.

**Error messages produced**
`Error: <geopm> Invalid argument: PlatformIOImp::push_signal(): domain_idx is out of range: at src/PlatformIO.cpp:407`

Specifically, this happened with `push_signal(GPU_ENERGY, 8, -1)`, when Reporter.cpp pushed GPU_ENERGY, board, 0 and platformIO attempted to use missing topology information to push each GPU domain index.

dannosliwcd, 2022-12-20:
Expected behavior is flexible to me, but I would like for there to be an easy path forward for users.

For example, we could try..catch on the push_signal in the reporter. If we hit the catch block, then either omit the failed signal from reports, or include it in reports, but with a NaN value.

bgeltz, 2023-08-23:
From Daniel:

Looks like Reporter.cpp still doesn't do any error handling on its call to push_signal.
The GPU signals are pushed to that list if it appears the signal name exists in any IOGroup. But GPU topology info comes from a separate interface than GPU signal info, and they can fail independent of each other.

Topology gets used in PIO's push_signal --> push_signal_convert_domain --> domain_nested call path. Topo populates domain_nested directly from domain_idx, which uses c-style return values to indicate errors. So if anything goes wrong inside domain_idx, topo tells pio to aggregate index -1 when Reporter asks for a board-aggregated signal.

Maybe 2 things remain: (1) The c++ domain_idx function should probably throw instead of adding -1 to the index set. (2) The reporter should probably make errors non-fatal when they occur on optional signals in the report (e.g., display a NAN or log warning).

The bug that made this a likely issue to encounter was fixed in https://github.com/geopm/geopm/pull/2842/files. So I think this is still an open issue, but a low-priority one now.

closed 2025-01-29 by bgeltz