geopm/geopm #2846: Unclear error message when writing GPU_CORE_FREQUENCY_MAX_CONTROL below GPU_CORE_FREQUENCY_MIN_CONTROL

asmaalrawi, 2023-02-13:
**Describe the bug**
I tried to change the gpu frequency by writing GPU_CORE_FREQUENCY_MAX_CONTROL and neglecting to write GPU_CORE_FREQUENCY_MIN_CONTROL. When doing so within an agent, I expect an error message that indicates which control was not written successfully. Instead, I observed the following error message: 

`Error: <geopm> Runtime error: BatchClient::write_batch The server is unresponsive: at src/BatchClient.cpp:96`

On the command line, when writing GPU_CORE_FREQUENCY_MAX_CONTROL via geopmwrite, the error message is also undescriptive:

```
> geopmwrite GPU_CORE_FREQUENCY_MAX_CONTROL board 0 1.0e9

Error: cannot write control: <geopm> Invalid argument: PlatformIOImp::write_control(): no support for control name "GPU_CORE_FREQUENCY_MAX_CONTROL" and domain type "10"
The following errors were observed:<geopm> Runtime error: SDBus: Failed to call sd-bus function sd_bus_call_method(), error:-5 name: not.known.Error.RuntimeError: geopm_pio_write_control() failed: <geopm> Invalid argument: PlatformIOImp::write_control(): no support for control name "GPU_CORE_FREQUENCY_MAX_CONTROL" and domain type "10"
The following errors were observed:<geopm> Runtime error: LevelZero::frequency_control: Sysman failed to set freq: at src/SDBus.cpp:32
: at src/PlatformIO.cpp:832
```

**GEOPM version**
2.0.0+dev346g72bd6b69

**Expected behavior**
A clear error message, such as Error: LevelZero::frequency_control: Cannot set max frequency below min frequency.

**Actual behavior**
An undescriptive error message regarding BatchClient, when attempting to write the control via the agent, and an error message regarding lack of control support on the domain type when attempting to write the control via geopmwrite on command line.

**Error messages produced**
Debug messaging was not enabled.

lhlawson, 2023-02-13:
Without the service there is error messaging indicating that an INVALID Argument is provided.
```
# geopmwrite GPU_CORE_FREQUENCY_MAX_CONTROL board 0 1.0e9
Error: cannot write control: <geopm> Invalid argument: PlatformIOImp::write_control(): no support for control name "GPU_CORE_FREQUENCY_MAX_CONTROL" and domain type "10"
The following errors were observed:<geopm> Runtime error: LevelZero::frequency_control: **Sysman failed to set frequency. Level Zero Error: ZE_RESULT_ERROR_INVALID_ARGUMENT**: at src/LevelZero.cpp:906
: at src/PlatformIO.cpp:832
```

With service (as user) the entire error message is not provided.
```
> geopmwrite GPU_CORE_FREQUENCY_MAX_CONTROL board 0 1.0e9
Error: cannot write control: <geopm> Invalid argument: PlatformIOImp::write_control(): no support for control name "GPU_CORE_FREQUENCY_MAX_CONTROL" and domain type "10"
The following errors were observed:<geopm> Runtime error: SDBus: Failed to call sd-bus function sd_bus_call_method(), error:-5 name: not.known.Error.RuntimeError: geopm_pio_write_control() failed: <geopm> Invalid argument: PlatformIOImp::write_control(): no support for control name "GPU_CORE_FREQUENCY_MAX_CONTROL" and domain type "10"
The following errors were observed:<geopm> Runtime error: LevelZero::frequency_control: Sysman failed to set freq: at src/SDBus.cpp:32
: at src/PlatformIO.cpp:832
```

cmcantalupo, 2023-08-04:
Duplicate of
https://github.com/geopm/geopm/issues/2887

bgeltz, 2023-09-18:
After discussion, the preferred solution here is to a retry loop in the write_batch() of LevelZeroIOGroup similar to the logic that exists in the NVMLIOGroup.  The goal is to retry setting controls if anything fails.

closed 2023-09-25 by bgeltz