geopm/geopm #2847: As a user, I would like to be able to set a fixed GPU frequency via a single control

asmaalrawi, 2023-02-13:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I need to write both GPU_CORE_FREQUENCY_MAX_CONTROL and GPU_CORE_FREQUENCY_MIN_CONTROL in order to set a fixed frequency and also have to remember to include the logic regarding the order of operations when using non-batch writes.

**Describe the solution you'd like**
I would like an additional alias control GPU_CORE_FREQUENCY_CONTROL to write both max and min control under the hood and perform those operations in the necessary order based on the frequency to be set and the existing values in those registers. The logic should avoid any circumstance when the contents of the max register are less than the contents in the min register. 

**Describe alternatives you've considered**
Alternative solutions:
* User should just always remember to write two separate registers and determine which order in which to write them
* Error messaging should be clearer when user fails to remember to write the two registers or makes a mistake with regards to the order in which they are written

**Additional context**
This register existed previously and was removed in order to clarify behavior on CPU, on which max frequency settings can have different effects based on the underlying mechanism used to steer frequency (HWP/acpi and frequency governor under the hood).

lhlawson, 2023-02-13:
Without the service there is error messaging indicating that an INVALID Argument is provided.
```
# geopmwrite GPU_CORE_FREQUENCY_MAX_CONTROL board 0 1.0e9
Error: cannot write control: <geopm> Invalid argument: PlatformIOImp::write_control(): no support for control name "GPU_CORE_FREQUENCY_MAX_CONTROL" and domain type "10"
The following errors were observed:<geopm> Runtime error: LevelZero::frequency_control: **Sysman failed to set frequency. Level Zero Error: ZE_RESULT_ERROR_INVALID_ARGUMENT**: at src/LevelZero.cpp:906
: at src/PlatformIO.cpp:832
```

With service (as user)
```
> geopmwrite GPU_CORE_FREQUENCY_MAX_CONTROL board 0 1.0e9
Error: cannot write control: <geopm> Invalid argument: PlatformIOImp::write_control(): no support for control name "GPU_CORE_FREQUENCY_MAX_CONTROL" and domain type "10"
The following errors were observed:<geopm> Runtime error: SDBus: Failed to call sd-bus function sd_bus_call_method(), error:-5 name: not.known.Error.RuntimeError: geopm_pio_write_control() failed: <geopm> Invalid argument: PlatformIOImp::write_control(): no support for control name "GPU_CORE_FREQUENCY_MAX_CONTROL" and domain type "10"
The following errors were observed:<geopm> Runtime error: LevelZero::frequency_control: Sysman failed to set freq: at src/SDBus.cpp:32
: at src/PlatformIO.cpp:832
```

cmcantalupo, 2023-07-19:
Is this a bug.  How does push_control(), adjust(), write_batch() work if the frequency goes both up and down?

lhlawson, 2023-07-25:
> Is this a bug? How does push_control(), adjust(), write_batch() work if the frequency goes both up and down?
Retry logic was added to the GPU IOGroups to handle the case where min > max fails and re-order the writes, but this requires both min & max controls being pushed.