geopm/geopm #2883: Control lock issue: contention for write lock

avilcheslopez, 2023-03-01:
**Describe the bug**
Reproducer: open a terminal, write control, close terminal, open a new terminal, attempt writing a control (all within a second)

**BUG DETAILS PENDING**

I tried to [...] and I expected [...] instead [...]

**GEOPM version**
Output from `geopmread --version`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Actual behavior**
A clear and concise description of what actually happened.

**Error messages produced**
Any output to standard error or standard output that may be related to
the bug.  Was debug messaging was enabled at build time?

_Note_:

If possible use the `--enable-debug` configuration option when
building GEOPM.  This will produce more verbose error messaging at run
time. To enable this when using the build script in the base directory
run the following command:

```bash
GEOPM_GLOBAL_CONFIG_OPTIONS="--enable-debug" ./build.sh
```

**Additional context**
Add any other context about the problem here.

_Note_:

Some helpful context could be the architecture of the system under
test, and other hardware or device driver details.  For bugs related
to the GEOPM runtime, the number of compute nodes, MPI ranks, and
OpenMP threads used by the application may be helpful context.

bgeltz, 2023-03-04:
From Lowren:
> Basically a large number of reads/writes back to back in a loop, with a workload run (LAMMPS in this case, NEKBONE would work too)

It has nothing to do with opening/closing terminals.

lhlawson, 2023-04-25:
See #2929

closed 2023-06-05 by bgeltz